opentx/opentx #2084: Can't add new input with Lua model.insertInput()

projectkk2glider, 2015-02-16:
Test script:
```lua
local function run(event)
  local ivalue={name="test", source=MIXSRC_Thr, weight=55, offset=3}
  model.insertInput(6, 0, ivalue)
  return 1
end

return {run=run}
```

projectkk2glider, 2015-02-16:
It actually crashes the simu!

jermoo, 2015-02-16:
exactly, change the command to model.insertInput(0, 1, ivalue) and it works. I'm looking at lua_api.cpp to see if I can figure it out but it's on the edges of my knowledge.. if I had the time I am very tempted to learn more so I can help/do the fixes.
Jer

jermoo, 2015-02-16:
I'm not able to get these to work either and they are part of the insertInput code.
  int first = getFirstInput(chn);
  int count = getInputsCountFromFirst(chn, first);

I can see the value of finding out the current inputs and simply adding the new one to the end, however, if we let the scripter worry about that then we can potentially use the insert command to update/overwrite an existing input line without the extra step of delete. 
I believe this extra code may be part of the issue.. still working to understand it all.
Jer

projectkk2glider, 2015-02-16:
When this issue is fixed, this is how you add the new input line at the end of existing input lines:
* first get number of existing lines for input 4: `noLines = model.getInputsCount(3)` (remember input number is zero based so 3 == Input4)
* use that number to add it to the end of the list: `model.insertInput(3, noLines, inputDefinition)` (again line number is zero based so if noLines =  2 ,then input is created on third line)

projectkk2glider, 2015-02-16:
My script to test this issue:
```lua

local function addInput(input, configuraton)
  local noInputs = model.getInputsCount(input)
  model.insertInput(input, noInputs, configuraton)
end

local function listInputs()
  for n = 0, 10, 1 do
    local noInputs = model.getInputsCount(n)
    print("input["..n.."] has "..noInputs.." lines")
    if noInputs > 0 then
      for m = 0, noInputs-1, 1 do
        local input = model.getInput(n, m)
        print("\tline "..m..": src="..input.source.." wt="..input.weight.." name="..input.name)
      end
    end
  end
end

local function deleteLastInputLine(input)
  local noInputs = model.getInputsCount(input)
  if noInputs > 0 then
    model.deleteInput(input, noInputs-1)
  end
end

local function run(event)

  print("Before:")
  listInputs()

  -- add inputs
  addInput(7, {name="test1", source=MIXSRC_Thr, weight=56, offset=3})
  addInput(1, {name="one", source=MIXSRC_Rud, weight=-30})
  addInput(100, {name="one", source=MIXSRC_Rud, weight=-30})  -- invalid input number, should do nothing

  print ("AFTER:")
  listInputs()

  -- delete one input
  deleteLastInputLine(7)
  print ("after delete:")
  listInputs()

  return 1
end

return {run=run}
```

@jermoo look at function `addInput()` in above script.

jermoo, 2015-02-16:
What If I don't want the input to be the next on the list? Just asking as I commonly use an input above 16 for CCPM heli setup for the Pitch channel. I'm playing withing using the lower values just for simplicity.

we also need access to the curve menu, is that currently possible? something like   addInput(7, {name="test1", source=MIXSRC_Thr, weight=56, offset=3, curve=expo(gv1)}) ..
or curve= any of the types with a data input that's correct for the curve datatype.. I'd rather input it at input creation time than seperately ....also need a way to edit inputs rather than create a new one.. seems wasteful to create a new one just to edit fields.
Thanks in advance.
Jer

jermoo, 2015-02-16:
ty for the script by the way.

jermoo, 2015-02-16:
oh..and the insertInputs is supposed to support a swtich assignment as well.

projectkk2glider, 2015-02-16:
>What If I don't want the input to be the next on the list? Just asking as I commonly use an input above 16 for CCPM heli setup for the Pitch channel. I'm playing withing using the lower values just for simplicity.

Where do you want it to be then? If you have an empty input line you can still use my function `addInput()`.

To edit existing input, you first have to delete it and then add a new one in its place. 

The curves must be created in separate step, because they can also be used in mixes. And there is currently no way to set curves from Lua. This is requested by #1647.

projectkk2glider, 2015-02-16:
>oh..and the insertInputs is supposed to support a swtich assignment as well. 

Yes it does, but the `model.getInput()` is not returning switch, I will also fix this at the same time.

jermoo, 2015-02-16:
thank you sir! Let me know if I can help in any way

projectkk2glider, 2015-02-16:
Are you able to compile OpenTX yourself? If yes then you could take my branch and try it yourself.

jermoo, 2015-02-16:
I'm setting my system up to do that, it may take a few days as I have many things going on. Once I'm setup I'm happy to learn to do this. I have some very base coding knowledge. 
jer

projectkk2glider, 2015-02-17:
Just a reminder to myself: write gtests for this!

lshems, 2015-05-13:
In 2.17 model.insertInput() doesn't seem to work at all??

the Switch is returned on model.getInput, but changing it, or any other parameter, doesn't help.

what am I doing wrong?

local function modelInput(input,line,parameter,newValue)
  local value = model.getInput(input,line)
	value[parameter] =newValue
      model.insertInput(input,line,value)
      model.deleteInput(input,line+1)
  end

projectkk2glider, 2015-05-16:
@lshems it works for me. Using this test script in simulator:
```lua

local function addInput(input, configuraton)
  local noInputs = model.getInputsCount(input)
  model.insertInput(input, noInputs, configuraton)
end

local function listInputs()
  for n = 0, 10, 1 do
    local noInputs = model.getInputsCount(n)
    print("input["..n.."] has "..noInputs.." lines")
    if noInputs > 0 then
      for m = 0, noInputs-1, 1 do
        local input = model.getInput(n, m)
        print("\tline "..m..": src="..input.source.." wt="..input.weight.." name="..input.name.." switch="..input.switch)
      end
    end
  end
end

local function modelInput(input,line,parameter,newValue)
  local value = model.getInput(input,line)
  value[parameter] =newValue
  for key,value in pairs(value) do print(key,value) end
  model.insertInput(input,line,value)
  model.deleteInput(input,line+1)
end

local function run(event)

  print("Before:")
  listInputs()

  -- add inputs
  addInput(1, {name="one", source=MIXSRC_Rud, weight=-30})

  print ("AFTER:")
  listInputs()

  -- test modelInput() function
  modelInput(1, 0, "switch", 2)
  modelInput(1, 0, "name", "mod")
  modelInput(1, 0, "source", MIXSRC_Ail)
  modelInput(1, 0, "weight", 66)
  print ("AFTER:modelInput")
  listInputs()

  return 1
end

return {run=run}
```

I get results:
```
Before:
input[0] has 1 lines
	line 0: src=77 wt=100 name= switch=0
input[1] has 1 lines
	line 0: src=78 wt=100 name= switch=0
input[2] has 1 lines
	line 0: src=76 wt=100 name= switch=0
input[3] has 1 lines
	line 0: src=75 wt=100 name= switch=0
input[4] has 0 lines
input[5] has 0 lines
input[6] has 0 lines
input[7] has 0 lines
input[8] has 0 lines
input[9] has 0 lines
input[10] has 0 lines
AFTER:
input[0] has 1 lines
	line 0: src=77 wt=100 name= switch=0
input[1] has 2 lines
	line 0: src=78 wt=100 name= switch=0
	line 1: src=75 wt=-30 name=one switch=0
input[2] has 1 lines
	line 0: src=76 wt=100 name= switch=0
input[3] has 1 lines
	line 0: src=75 wt=100 name= switch=0
input[4] has 0 lines
input[5] has 0 lines
input[6] has 0 lines
input[7] has 0 lines
input[8] has 0 lines
input[9] has 0 lines
input[10] has 0 lines
source	78
switch	2
weight	100
offset	0
name	
source	78
switch	2
weight	100
offset	0
name	mod
source	78
switch	2
weight	100
offset	0
name	mod
source	78
switch	2
weight	66
offset	0
name	mod
AFTER:modelInput
input[0] has 1 lines
	line 0: src=77 wt=100 name= switch=0
input[1] has 2 lines
	line 0: src=78 wt=66 name=mod switch=2
	line 1: src=75 wt=-30 name=one switch=0
input[2] has 1 lines
	line 0: src=76 wt=100 name= switch=0
input[3] has 1 lines
	line 0: src=75 wt=100 name= switch=0
input[4] has 0 lines
input[5] has 0 lines
input[6] has 0 lines
input[7] has 0 lines
input[8] has 0 lines
input[9] has 0 lines
input[10] has 0 lines
Script finished with status 1
```

The input that is modified with your function ends up as expected:
```
input[1] has 2 lines
    line 0: src=78 wt=66 name=mod switch=2
```

lshems, 2015-05-16:
Thanks. Two things:

I found it stops working when having lots of inputs. But before the maximum
allowed is reached. It is the unpredictable way this happened I worry about.

How do you get the print command to work?

projectkk2glider, 2015-05-17:
>I found it stops working when having lots of inputs. But before the maximum allowed is reached. It is the unpredictable way this happened I worry about. 

You are right, we have a bug here.

PR #2086 (2015-02-16): Projectkk2glider/issue 2084 lua insert input
Closes #2084

Script for testing:
```lua

local function addInput(input, configuraton)
  local noInputs = model.getInputsCount(input)
  model.insertInput(input, noInputs, configuraton)
end

local function listInputs()
  for n = 0, 10, 1 do
    local noInputs = model.getInputsCount(n)
    print("input["..n.."] has "..noInputs.." lines")
    if noInputs > 0 then
      for m = 0, noInputs-1, 1 do
        local input = model.getInput(n, m)
        print("\tline "..m..": src="..input.source.." wt="..input.weight.." name="..input.name)
      end
    end
  [... cut]
commit 23c9efa5 (2015-02-16): Fixes #2084: index changed to unsigned for Lua mixer funcitons
commit f261118a (2015-02-16): Fixes #2084: Lua model.insertInput() could not add a line to an empty input or at the end of the existing input lines
commit 2b8374f4 (2015-02-18): Fixes #2084: Lua model.insertInput() could not add a line to an empty input or at the end of the existing input lines
Also added unsigned to various vars (ported from master)
closed 2015-05-17 by projectkk2glider