opentx/opentx #2085: Mix delay slow function only works on first actuation

u2fletch, 2015-02-16:
I have a throttle mix which will activates full throttle  with a 3s "slow" after a 3s "delay" when you select the SD switch up.  Used for hand launching a flying wing.  The switch delay works okay once, then after that, any subsequent use there is no "slow".  The delay still works, but the change becomes instantaneous.

hrenz, 2015-02-16:
No, slow works well.

Slow is allways new triggered when the Mixer source changes!
SD up and Sd down is allways the same Source the SD
the Trick ist take a log switch instead a switch

L1  as source 1
L2  as source 2

so allways  the slow will trigger new, and it works fine.

kilrah, 2015-02-16:
Slow works on source value changes, not on line activation/deactivation.

closed 2015-02-16 by kilrah