opentx/opentx #2090: Companion: Minor cosmetic bug with play track

JoeB99, 2015-02-18:
Cosmetic bug: If you have 2 play track lines in special functions, and you click the first to play, then click the second before the first has finished playing, the first stop icon does not change back to play.

To reproduce: 
Click play on the first SFx 'play track'
Click play on the 2nd SFx 'play track' BEFORE the first has finished.

This is cosmetic and does not effect the functionality of the software.

jtaylor2, 2015-03-30:
@projectkk2glider I wanted to do a pull request for a fix for this but messed up and evidently pushed it directly to next.  Should I back it out or leave it?

kilrah, 2015-03-30:
It's good like that, just try to remember to create a branch in the future :)

You can do the same change on master.

closed 2015-03-30 by kilrah

projectkk2glider, 2015-05-02:
@jtaylor2 slipped a bug into master 730bc8c608529fdd1f6ce6b092c4334c2741538b and next 459837d6f36502ca97cf44e057304f3779eb92d1 . Jim use a pull request next time :wink: