opentx/opentx #2091: [Bug][Companion 2.0.15]Various bug on model wizzard

Sacre100, 2015-02-18:
Model wizzard allows only to select Channel 1 to 8 on Taranis instead even if Channel 1 to 32 are available.

[image]

On "Tail" window, when Channel 1 to 8 are used for other control, the dropdown propose only Channel 1 and "Next" button has no effect (only "Prev" or "Cancel" buttons have an effect)

[image]

griebd, 2015-03-13:
We have two different things here....

1. A bug: when no channel left to be used on 'Tail', or anywhere in the wizard, a error message should be shown. (I will look at that... :wink:)

2. A project choice to be made: keep the actual wizard, made for a 8 channel (maximum) model. Or change the wizard so that the user chooses the number of channels for the model.
I simple don't use the wizard and I think (made wrongly) that if you need more than 8 channels you don't need a wizard... :wink:
Anyway, if it's decided to do the change, I'll look at it too.

kilrah, 2015-03-13:
> I think that if you need more than 8 channels you don't need a wizard... :wink:

Same here, nearly closed the issue straight away because of that. The wizard has been done as "basic handholding for the first steps", so you can see at how it did what you told it to... if you need everything you've already outgrown it.

griebd, 2015-03-14:
Did a fix to show a message on "tail" page when there is only one channel left.
As for the number of channels, I still think it's not necessary. If you need "that power", OpenTX IS for you!!! The wizard is only a introduction to it's "power", as pointed by @kilrah.

griebd, 2015-03-17:
Already merged for 2.0.16.

commit 08f70bc6 (2015-03-17): Merge pull request #2143 from griebd/model_wizard_2091

Fix #2091
commit 4b510153 (2015-03-17): Fix #2091
commit 56bc0d81 (2015-08-16): Fix #2091
closed 2015-03-17 by griebd