opentx/opentx #2092: [Bug] Cannot adjust 2 GVs with trim button

LapinFou, 2015-02-18:
There is wonderful, very useful function in OpenTX: the "Adjust GVx" function.
Even better if a trim is used as a source (but also used for "normal" trim operation), then OpenTX will temporarily unmaps the associated trim value.
Ex:
SF1 SC^ Adjust GV1 TrmT

You are the best guys !!
:)

However this doesn't work well with 2 special functions.
ex:
SF1 SC^ Adjust GV1 TrmT
SF2 SC^ Adjust GV2 TrmT

Only GV2 is update, not GV1.
[image]

projectkk2glider, 2015-02-18:
Confirmed.

projectkk2glider, 2015-02-28:
Hmm, I wonder if this is really needed!?  

The thing is this would require a big changes in source code. How useful it really is to change two or more GVARS at the same time with the same trim switch? You can use two work-arrounds:
* either use different trim switch for each gvar
* or use different condition for each gvar so that both are not active at the same time

I any case this is not a but, but (missing) feature.

LapinFou, 2015-02-28:
I don't really agree about you conclusion. It is more a bug than a missing feature.
Your answer remembers me a marketing guy in my previous job.  ;-)
Anyway, I'm not myself using this method to update 2 GVAR. But, a friend try to do that on a complex sailplane and reported the bug to me SO, I opened a ticket. There is a couple of possible workaround.
So it hope to you.
:)

bsongis, 2015-02-28:
up to you ;)

LapinFou, 2015-03-01:
Oups !!
It is up to you !  :)

Sacre100, 2015-03-01:
As GVARS are incremented or decremented by the trim and not assigned to the trim value, GVARS can have different value before and after clicks on the trim. 

On the exemple here under, the expected result is : 
- GV1 updated when SC-
- GV1 and GV2 updated when SC^

[image]

Currently, either GV1 or GV2 are updated but not both when SC^

However, I don't know if this is a real need, it's just to have a consistent behavior.

kilrah, 2015-03-01:
When you enable an "Adjust GVAR" SF with a trim as source the trim switches to editing an alternate register dedicated to that GVAR.
There would be a conflict if wanting to do more than one at a time, which is why @projectkk2glider mentioned major changes would be needed.

Sacre100, 2015-03-01:
I understand that's a major change, I'm not stupid, that's why I said "it's JUST to have a consistent behavior."

LapinFou, 2015-03-01:
I understand. I can leave as it is, because it is already a so great feature. :)

To avoid any setup trouble, maybe it could be nice to insert some check in Companion/OpenTX which will forbid the possibility to create a non-working configuration as reported in this ticket? I guess, it is not easy to do.

PS: Despite this discussion, are you OK to rename the trims? As explained at the end of the 1st message? This is easy on should avoid confusion between INPUTS and SPECIAL FUNCTIONS (same name, but not same meaning).

kilrah, 2015-03-03:
> By the way, on order to avoid any confusing, maybe the source trim should name "xTrim Down/Up" instead "xTrim".

No, because firstly Trmx up/down is already used for the trim keys, and secondly all you've got is 4 characters. If you find something... but it will likely be even more confusing than something pretty much everybody understands by now.

LapinFou, 2015-03-04:
OK. If there is only 4 chars, the best solution has been already used.
According all the explanations, maybe this can be changes for OpenTX 2.1 or 2.2. Otherwise, I thin you may closed this ticket.

closed 2015-03-30 by kilrah