opentx/opentx #2094: Taranis FW update to 2.0.15 has silenced "minute" audio call out on Timer

GarySeven, 2015-02-18:
Updated FW on my Taranis to "opentx-taranis-2.0.15" from 2.0.13.  Now, even though "Timer 1" in "Model Setup" has not changed, I no longer receive audio callouts each minute after being initiated by throttle movement.  Settings are:  Timer 1 / Tht  00.00 / Persistent Off / Minute Call (box checked) /Countdown  Voice

Audio works fine when Taranis is switched on.  Also, all switches I have assigned audio in "Special Functions" work as they should.

projectkk2glider, 2015-02-18:
Already fixed, will be available in 2.0.16, duplicate of  #2009.

closed 2015-02-18 by projectkk2glider

GarySeven, 2015-02-18:
Thank you.  I looked high and low before I posted because I did NOT want to duplicate issues and add to "the fog."  Sorry I missed it, but thanks for the quick reply.