opentx/opentx #2095: Request current Flight Mode be made accessible

mpjf01, 2015-02-19:
It would be useful to be able to operate a switch on the radio to play a track announcing the current Flight Mode. At present doing this requires a combination of GVs with Logical Switches and Special functions for each FM. 

It would be simpler if the value of the currently operating FM could be made available and easily used to play the same track as is currently played when the FM changes (Such as Launch-On for eg.).

projectkk2glider, 2015-02-19:
Interesting idea.

mpjf01, 2015-08-06:
I appreciate that the developers are busy with important stuff but I would like to put in another bid to have this enhancement added please. I fly full house gliders a lot with 6 and seven flight modes. At present the only ways to tell which mode I am in is to remember where I am in the sequence, to look at my Taranis screen or to switch to another mode and then back again. In the heat of competition I don't remember well, having to take my eyes off the model when it is far away is dangerous and changing FMs to something I don't want at the time can have deleterious effects. It really would be useful to be able to play a track to announce the current FM at any time. Please help.

kilrah, 2015-08-06:
In the meantime... all you need is one LS and one SF per mode, no GVARS or stuff.

mpjf01, 2015-08-06:
I have enough LS left to do that. Can I access the FM sounds that are under Sounds/en/model name? Or do I have to copy these sounds to Sounds/en?

mpjf01, 2015-08-06:
If I copy them to sounds/en they don't appear as a play track option.

projectkk2glider, 2015-08-06:
>If I copy them to sounds/en they don't appear as a play track option.

Name too long? Max 8 chars + extension.

fraternl, 2015-08-06:
If you make logical switches

LS1 AND FM1 SH 
LS2 AND FM2 SH
LS3 AND FM3 SH

LS1 play track FM1  
LS2 play track FM2

Or am I missing something?

mpjf01, 2015-08-07:
I have no tracks titled FM1, FM2 and so on. The normal tracks played when FM changes are in Sounds/en/model name. These don't appear to be accessible from a SF - as I said above. To see them I had to rename each one and copy to Sounds/en. This has to be done separately (if the sounds are different) for each model. 

It is certainly possible to do this and to create (in my case) an extra 6 LS and SF lines for each of my 5 models, but I'm hoping that someday the process will be facilitated by a firmware enhancement.

fraternl, 2015-08-07:
That's part of the Amber soundpack
I assumed every advanced user has it installed...

These tracks are in /sounds/en/

I know nothing about sound tracks in models

mpjf01, 2015-08-07:
You are right, no change to the firmware is required, it would just make things a lot easier. There's many areas of functionality that have been added to the firmware for this reason in the past. Flight modes themselves are one example. Just because you can implement a function with existing firmware doesn't mean that opportunities to do it more easily should not be taken up.

As regards sound tracks in models, they are the sounds that can be set up to be played automatically on FM change. I assumed every advanced user would know that.

I have had the Amber sound pack, and it's updates, installed since it became available, my version has no sounds titled FM1, FM2 etc as I said, and even if it did it wouldn't matter to me in this case. I have made my own custom sounds for the FM changes as the Amber pack does not support my requirements in this particular respect. 

I'm not sure why you feel the need to contribute to this discussion. You clearly don't use this functionality - if you did you would have a different view - and suggesting that my request is indicative of a lack of creativity can only be unhelpful, even offensive.

fraternl, 2015-08-07:
In my first post in this thread I left open that I could be missing something. 
I even took the effort to write an example how to do what I thought you were trying to achieve

I totally agree with the first paragraph of your last post and I think you should have left it at that.

I did appreciate your pun and it put a smile on my face...

The last part is totally out of line.

I don't know you and you don't know me. I'm not here to troll and play down your request and hope for you it will get some love of the devs and make things a bit structured for you.

closed 2020-07-08 by stale[bot]