opentx/opentx #2096: Feature request: Channel lock special function

TexasFPV, 2015-02-19:
Using the pots on the Taranis is useful for tuning multirotor gains, but after the gains are set, the pots can be accidentally moved.  It would be useful to "lock in" the current value of the pot by using a toggle.    Whatever channel is using the pot will remain at the value read from the pot when the lock is toggled, regardless of the pot position.  This would be similar to the current channel override special function, without a predetermined channel value.

davxlw, 2015-02-19:
Just add a mix on the channel you want to lock with itself as source and multiplex to replace.
Here is an example with pot S1 controlling channel 6:
[image]

lepicfoto, 2015-02-20:
nice i was looking for something just like that, tnx

kilrah, 2015-02-21:
Or use trims / global vars to set them.

closed 2015-02-21 by kilrah