opentx/opentx #2097: A radio-wide calibration mode to assist calibration of servo's

fraternl, 2015-02-20:
In my models I'm using the calibration mode described on this page
http://rc-soar.com/opentx/setups/calmode/calmode.htm

This means I have to create these mixes for each model, create the logical switch for it and a special function to warn me it is in effect.
Due to the open nature of OpenTX (no input and channel assignment by default) it is not that easy to implement radio-wide.
At this moment I have no idea to implement it radio-wide.

I do know that I would like to get it out of my model and mixes, so it would become a more general thing.

projectkk2glider, 2015-02-28:
Calibration mode described in above link (I also use it) is thigtly coupled with the specific model and can't be generalized. 

There is similar idea already open at #173

fraternl, 2015-03-02:
@projectkk2glider  I wouldn't know either how to do it.

If there would be a radio-wide set of logical switches and a radio wide set of special functions it would only need the lines in the mixes.

I assume there's already a feature request for universal (as the word "global" is already taken for modelwide) vars, switches and special functions?

jermoo, 2015-03-03:
To implement this radio wide a few ways could be used now. The easiest way would be to add it into a model and use that model as a template for all future models. It could also be added to the model wizard if you use that. Another easy option would be to write a function Lua. Being that it's one line being added to each channel mix it should be trivial. My personal choice would be to write a function lua that I can run on demand only when I want it. I use a similar setup on my models and delete it once the model is flight ready. Just my opinion.

kilrah, 2015-03-03:
I believe the lua solution already exists. Someone made a script that when run adds the required entries if not present, and removes them if present.

fraternl, 2015-03-04:
Ah, that's interesting....
I will investigate it.

closed 2016-11-27 by fraternl