opentx/opentx #2100: Feature Request: sbus servo channel changer

pinkywafer, 2015-02-20:
Would it be possible to add functionality of servo channel changer using module port (or maybe trainer port)

projectkk2glider, 2015-02-20:
Maybe this could be also done in #1599.

projectkk2glider, 2015-03-09:
@bsongis, @kilrah, @MikeBland Do we actually know the protocol for this?

kilrah, 2015-03-10:
I don't. FrSky do, maybe Mike asked them.

MikeBland, 2015-03-13:
I have no idea of the protocol, never asked.

LapinFou, 2018-01-14:
Some news: FrSky release a new product named SD1 → https://www.frsky-rc.com/frsky-sbus-decoder-sd1/
When downloading the firmware, you got some included LUA script to change the channel number.
Maybe this can help?
Firmware can be downloaded here: https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/SD1/FW-SD1-171114%20and%20lua.zip

closed 2020-07-08 by stale[bot]