opentx/opentx #2107: Feature: Modulo (%) operator for logical switches

JoeB99, 2015-02-23:
I had a brief discussion on the forums with a user that had a logical switch L1 for when the throttle was not -100 and he wanted the timer to call out the time when it was less than zero, every 10 seconds.
This was achieved my adding:

L1 = throttle > -100
L2 = timer < 00:00:00 AND L1
SF1 = L2, play value, timer 1, 10 secs

The problem is when the L1 condition is false, the 10 second callout resets (correctly).

Therefore, it is not possible to have the timer callout at -10secs, -20secs etc. It calls out 10 secs from when L1 becomes true again.
This can only be achieved with a modulo operator %

L1 = throttle > -100 (throttle is on)
L2 - timer < 00:00:00 AND L1 (timer less than zero and throttle is on)
L3 = (10 MOD timer) AND L2 (timer is a multiple of 10 and less than zero, and throttle is on)
SF1 = L3, play value, timer 1, 10 secs

hrenz, 2015-02-23:
Hy

you can do it with the Flipflop Funktion
like this one

[image]
[image]

hrenz, 2015-02-23:
or this one
[image]
[image]

JoeB99, 2015-02-23:
With the above, when i lower the throttle for 3 seconds, then raise again, the count no longer calls at timer MOD 10 seconds.

A MOD function would be much neater and easier to understand.

fraternl, 2015-04-08:
Oh,,,,, I just thought of using a modulo for calling the consumption of my battery each 300 mAh....
So, it seems it's not yet in there...
I believe it is in LUA, but it's much neater to have it in the radio as well...

fraternl, 2015-04-08:
I would like that mod function too, but I noticed the other functions are only taking 2 values. With a MOD function you want it to be true if the result would be some integer.
Most of the time you would want it to be 0, but now I'm only thinking of my application.

Would an "=INT" function be enough for you if it would make it easier to implement?
BTW...  I don't know if that's true or not....

fraternl, 2015-04-08:
http://openrcforums.com/forum/viewtopic.php?f=45&t=6907&p=95533#p95533

zieren, 2015-04-14:
I don't think the |d|>=x function is quite as good as modulo. For example, I would like to announce altitude every 20m (or time, or voltage...). The delta function doesn't guarantee that all announcements are at n*20m. For example, if the initial value is 2m then the next announcement will be "22m". Or say the initial cell voltage is 4.18V and I want to announce every 0.1V - the next announcement would be 4.08V.

I've set it up with the delta function at the moment, but it's definitely distracting to have to parse "seventy-five" or "ninety-six" etc. rather than just the multiples of the modulo.

bsongis, 2015-04-20:
Last commit for discussion, not sure it will fullfill the requirements ...

zieren, 2015-04-20:
Awesome, thanks! Just to clarify, the removal of LS_FUNC_RANGE is a cleanup, or are you removing that function?

I've added some comments, let me know if I can help with translations. Some colleagues at work are native Russian, Hungarian, English and Polish speakers and would probably enjoy this :-)

projectkk2glider, 2015-04-20:
One observation from me: the new modulo function requires **exact multiple** to trigger. This might not work as users want/expect. For example if we set modulo to 30 and we get following telemetry values: 57, 59, 61, .. the modulo function won't fire (the 60 value is never seen by the telemetry).

bsongis, 2015-04-20:
Yes, it's always the same with == there!

zieren, 2015-04-20:
So how about using the same mechanism as for a delta increase/decrease? Like "|d|>=1" for "d=int(x / y)".

That way there'd be a trigger when the value crosses from one bucket to another. For slowly changing values that would be at the excat bucket boundary, but when the boundary is skipped it would still work. It would also fix the problem with the existing delta function that offsets keep adding up.

JoeB99, 2015-04-20:
I believe the modulo function needs to be a true modulo, not an approximate modulo. 

Maybe there could be 2 separate modulo operators: 

1) exact modulo:
Timer needs to sound every 30 secs, on 1:30 2:00 2:30 etc

2) Approx Modulo with adjustable precision
Announce voltage every 0.3V +-0.02 precision: 
1.21V
1.53V (missed as out of range) 
1.78
1.82

That should cover most use cases....maybe! :)

kilrah, 2015-04-21:
IMO the new commit handles the "exact match", and the existing diff already serves the other case quite well.

zieren, 2015-04-21:
I think an exact modulo only works for values that are known to change slowly, such as time. For these we know that we will see a sequence that contains all values and no gaps (as JoeB99 points out).

However most telemetry data is not like that (see the voltage example; another would be altitude). If I want to announce altitude every 20m it's not possible with the modulo, not even with a precision argument. For example:

..., 18, 18.5, 19.9, 20.2, 20.3, 23, ...

With an exact module this would miss the 20. With a precision of e.g. 1m it would announce three values instead of one.

What would work is a combination with the existing delta semantics: What we want to know is when the next multiple of some value (20 in the example above) is reached or exceeded. That's "|d|>=1" for "d=int(x / y)".

This doesn't need an accuracy argument (so it's less confusing and easier to get right) and works for all cases.

fraternl, 2015-05-06:
I thought about using it this way:

LS1     a%x<y   Vfas  20   0.9

18 % 20 = 18  false
18.5 % 20 = 18.5 false
19.9 % 20 = 19.9 false
20.2 % 20 = 0.2   true
20.8 % 20 = 0.8    true
21 % 20 = 1     false

commit 601aa7d7 (2015-04-20): Fixes #2107 - Modulo logical switch added
closed 2019-11-15 by stale[bot]