opentx/opentx #2109: Geofencing?

mogplus8, 2015-02-24:
There's been a lot of talk around the clubs lately about the importance of remaining within the boundaries of club flying areas. I know with our club it's becoming crucial as houses are being built closer and closer to the flying field. So it occurred to me that it might be possible to use a GPS module and some clever software to allow the GPS coordinates of a flying field to be set in a transmitter, and have a warning sound when the model flies outside the bounds of the field. Maybe even if the model flies too far "out of bounds" it would be possible to have it autonomously turn around and head back to the "in bounds" area (temporarily ignoring the transmitter). This could also be a great model saver for those of us whose eyesight is not what it was. Just a thought.

kilrah, 2015-02-24:
Someone has already made a Lua scripts for out of gps bound detection, a short search should bring it up. 
Auto return would require an onboard flight controller.

mogplus8, 2015-02-24:
Thanks for the quick reply!
Is this the script you were referring to? It's the only one I've been able to find. http://rcsettings.com/index.php/viewdownload/13-lua-scripts/146-protected-area.
Thanks.

kilrah, 2015-02-24:
Yes I think that's the one.

closed 2015-02-24 by kilrah