opentx/opentx #2110: BUG: Model compare missing SF differences

JoeB99, 2015-02-25:
SF differences are missed in compare model. See attached image.

[image]

How do i upload an eepe here to show the issue?

projectkk2glider, 2015-02-25:
>How do i upload an eepe here to show the issue?

No need, the compare has more issues, it needs someone to take time and fix it thoroughly. But there are so many other issues more important...

Related #1862

romoloman, 2015-04-27:
This one has been fixed, also curve compare is back, but there is still a lot of stuff either in compare and in printmodel to be fixed.

closed 2015-04-27 by romoloman

JoeB99, 2015-04-27:
Great. I use compare all the time when changing settings at the field and then syncing them back to my master file at home, or when helping a friend set up models.

bsongis, 2015-04-27:
@romoloman thanks for taking care of those issues!