opentx/opentx #2113: Global variable seting menu missing in 9XR

tomekmgr, 2015-02-26:
In 9XR with M128, there is no menu to configure Global Variable in radio.
I can, use them, adjust in special functions, but there is no way to view or define a initial value in radio setup. 
It can be done only via Companion, but at the field it's not an option.

kilrah, 2015-02-27:
They are in flight modes.

closed 2015-02-27 by kilrah

tomekmgr, 2015-02-27:
In flight modes In Companion, but not in radio menu.
Maybe my description wasn't clear.

I would like to change / see Gloval Variables values in radio menu.
Such feature is described in openTx documentation for Taranis in section 'Global variables'.
Menu screen 9/12.

Is it possible to enable this option for 9XR?

kilrah, 2015-02-27:
On the radio it's at the bottom of the curves page.

tomekmgr, 2015-02-27:
Found it. Thank you. :)