opentx/opentx #2117: [Enhancement] Increase Failsafe security

LapinFou, 2015-03-01:
They are 4 possible failsafe mode:
[image]

I proposed to create a 5th one which will be the default one: an alarm (beep + pop-up message, the same as throttle warning) which will remind you to setup the desired failsafe mode each time you select this model.
To many people are forgetting the setup the failsafe mode. It is a shame, because it can save your model...
;)

bsongis, 2015-03-01:
Yes, why not!

LapinFou, 2015-03-04:
Glad you like the idea. :)

BarryCole, 2015-04-11:
Anything that helps with the failsafe settings is a good thing. The default of HOLD is a bad thing that most manufacturers use. It was partly the cause of a fatality in the UK some years ago.  Would it be possible to make the default Custom, with the defaults all set to -100, plus your reminder as above.

BC

kilrah, 2015-04-11:
Defaults at -100 is a horrible idea, depending on the rest of the configuration that can mean full power and/or break servos.
There is no good default. The only good solution is you remembering to set your failsafe...
But the default could at least be Receiver as we get people who read the receiver's manual which explains to use the button on it to set failsafe - not telling them that would be overridden by the radio's default settings.

BarryCole, 2015-04-11:
Yes, Klrah, I did not think that through before writing it. Last position hold is not a good thing though. It was the part cause of a fatality over here. The plane had just taken off, so had full power, and up elevator. Control was lost due to a frequency clash on 35Mhz. The plane did a full power loop, and hit a youg lad in the back of the head, killing him instantly.

Your custom system is brillint, if only we could come up with a way of getting everyone to use , and check it.

I still do not see many people checking their fail sfes settings on each outing. Such is life.  Keep uo the good work.

BC

BarryCole, 2015-04-11:
Yes, Kilrah, I did not think that through before writing it. Last position hold is not a good thing though. It was the part cause of a fatality over here. The plane had just taken off, so had full power, and up elevator. Control was lost due to a frequency clash on 35Mhz. The plane did a full power loop, and hit a youg lad in the back of the head, killing him instantly.

Your custom system is brillint, if only we could come up with a way of getting everyone to use , and check it.

I still do not see many people checking their fail sfes settings on each outing. Such is life.  Keep uo the good work.

BC

LapinFou, 2015-04-12:
It is the main reason why I opened this ticket. I think displaying a warning is the less worse solution.
Then it will be the user responsibility to take in account the message or to to bypass it.
:)

BarryCole, 2015-04-12:
Yes, I agree that does seem a great idea.

BC

romoloman, 2015-04-27:
I do not agree in having an alarm on the TX if failsafe is not set (or left to HOLD) every time I turn on the radio, maybe we can give an alert after we finish the bind procedure, but after then everything is left to the user.

Setting a failsafe is important but not for the reason written "To many peoples are forgetting to setup the failsafe mode. It is a shame, because it may save your model..." 
Failsafe is not there to save models but to make it inoffensive to others even if it means that the model will be damaged/destroyed.
As already written there is no way to properly set a failsafe without a deep knowledge of the flying characteristics of the model itself. The best way for a plane to try to save as much as possible of a model is the flat spin drop, but  you cannot set commands for flat spin until you flight the model (typically are: engine at 0, full ele, an amount of ail and an amount of rud)
On other models (multirotors, elicopters) failsafe settings are indeed different.
Also failsafe may be set on the RX ie this is the only way if you want no signal on failsafe.

bsongis, 2015-04-27:
I think I like the original idea: warn the pilot he has to setup a failsafe

LapinFou, 2015-04-28:
@romoloman : the idea is to warn the user to choose the appropriated failsafe mode corresponding to his needs. Then, the user can disable the warning on purpose.

commit 8cb65392 (2015-05-08): Fixes #2117
commit 2b18783a (2015-05-10): Fixes #2117
closed 2015-05-08 by bsongis