opentx/opentx #2119: [Enhancement] SpecialFunction editing order

JivaroFAD, 2015-03-01:
Edit SpecialFunction first then select Switch second : 
[image]
Switch is move at the right of the screen, right next to the activation check box, that seems more logical because we don't always want to use a switch (switch value can be "ON" if function is always activated).

jtaylor2, 2015-03-01:
In my opinion changing a user interface that thousands of people are already using and know how to use would need to provide significant improvement, and I fail to see how this suggestion would be an improvement or any more logical than the way it is now.

Special Functions can have a variable number of following fields with activation (switch) and action being the only two actually required in all cases (ie: L2 vario).  So having the activation (switch, ON, ONE, FM), whether you call it switch or something else, and action in the first two fields seems to me like the only logical choice.  Having switch as the first field makes it easy to temporally disable a SF without deleting it by just moving down and changing the first field to --- and  I also think that having switch, which is a narrower field, next to the SF number is more aesthetically pleasing. 

[image]

Jim

kilrah, 2015-03-01:
And even with an order change it would be needed to go to the now 3rd column to choose either the switch or ON.

JivaroFAD, 2015-03-01:
@jtaylor2 : when you create a special function, you have first in mind the function, not the associated switch, if exist.
@Klirah : why not using "ON" as default ?

jtaylor2, 2015-03-01:
@JivaroFAD :  It really doesn't matter which you have first in mind, you still have to program both the switch/activation and the function so why move one of them way over on the right side of the display instead of keeping them together next to the SF number?  And I'm sure many people are like me and use the same switch for the same function on all their models so perhaps think of the switch first (I have never even considered which I think of first).  For example I always put my motor cut on SF, Gear on SA, Flaps on SD, and rates on SB so which do I think of first.  And if I change a SF after setup it will probably be changing the switch on a line, not the function.  I always put my default functions on the UP position of a switch, so if I decide, for example, I want low rates to be the default instead of high rates I'll change the switch on the mix lines and also the switch on the SF play value lines that announce it.

I personally think the suggested change is less logical than the way it is now, but that's just me.  But even if it were a little more logical, would it be worth changing a gui screen that thousands of people already know and use?.

JivaroFAD, 2015-03-02:
My proposal refer to 9x and not Taranis...
On 9x / Next :  
- there's no "ON" state and there's an activation check box that is not on taranis.
- this check box is redundant with activation state

kilrah, 2015-03-02:
Sorry but the ON state is there and the checkbox is only there for the functions that require it for safety, just like on Taranis.

[image]
[image]

JivaroFAD, 2015-03-02:
My last flash with Next branch give only "OFF" state et not "ON".

kilrah, 2015-03-02:
ON should be at the opposite place in the list (+ side, if you have dblkeys option press + and -).

JivaroFAD, 2015-03-02:
Didn't find it, but I will check again this evening.

jtaylor2, 2015-03-02:
>My proposal refer to 9x and not Taranis...
On 9x / Next : 

It doesn't say that, I thought you were suggesting it for all platforms.  If I had known it was only for 9x I wouldn't have commented.  Suggest you add "for 9x" to the title.

JivaroFAD, 2015-03-02:
Verified states, from left to right : 
ID0, ..., TRN, tR1, .., tAr, REa, REb, L1, ..., L12, OFF, !L12, ..., !L1, !REb; !REa, !tAr, ..., !R1, !TRN, ..., !ID0
No "ON" :(

kilrah, 2015-03-03:
What options? Do you compile yourself or use the supplied nightly binaries? There is no RE on 9x unless explicitely enabled with the rotenc option, and that is very likely broken at the moment as that wouldn't have been taken care of in next yet so would be a bug more than anything. Next is barely in alpha stage. 

[image]
 
I feel we're all losing our time due to bad communication here.

bsongis, 2015-03-03:
@JivaroFAD I just fixed the overflow problem. Companion should also be inline with the MEGA2560 EEPROM format now.

JivaroFAD, 2015-03-04:
Thanks Bertrand :)

@Kilrah :  rotary encoder option on Mega2560 board (nearly the same as Gruvin9x).
Are you sure that rotary option could explain why there a "OFF" state and not a "ON" ?????

bsongis, 2015-03-04:
I have a ON since yesterday ;)

JivaroFAD, 2015-03-04:
That's very great ;)

closed 2015-04-20 by bsongis