opentx/opentx #2121: Complete set of Vars, Switches and special functions that are radio-wide instead of model-wide

fraternl, 2015-03-02:
A lot of switches, variables and special functions need to be duplicated for each model.
Having a separate (radio wide) set would make things more structured.
More advanced would be some template system to link in a group of sets that each contain a specific custom function.

By default it should of course work as it is working now.

kilrah, 2015-03-02:
Global SFs are already implemented in next.

closed 2015-03-02 by kilrah

mpjf01, 2015-03-03:
If "Global SFs" are for all models the nomenclature will be confusing. Does this mean that "Global Variables" will become "Model Variables" or just "Variables" or similar, with the term "Global Variables" used for the variables that apply to all models to be consistent with the new "Global SFs"?

fraternl, 2015-03-03:
AFAIK the terminology is only used for Variables, but that could be confusing too. "Universal" is a good candidate?
But what about "Universal Variables" and "Universal Logical Switches"?

Special Functions could react on normal "Logical Switches" and on "Universal switches"
The same goes for the switches reacting on "Global Variables" and on "Universal Variables"

I assume the amount of these switches are at least doubled...
I already ran out of switches on one model.

jermoo, 2015-03-03:
how about Radio Variable and Radio swtich..etc.. The term "Radio" being synonymous with the universal nature at the radio level rather than model level.

mpjf01, 2015-03-03:
That would be consistent with the existing use of "Radio" in Radio Setup on the radio (but not Companion). However it would seem that they have already been  called "Global Special Functions" as per post above.