opentx/opentx #2122: % in model name not saved

JoeB99, 2015-03-03:
Using companion, i named a model "MSR +125% o2" and the percent symbol is replaced with a space when the eepe is saved.

kilrah, 2015-03-07:
Special characters other than those you can enter on the radio (_ - . ,) aren't available in model names. Known limitation, was decided not to act on it but maybe we could also limit entry to supported chars in companion.

#651 is loosely related.

JoeB99, 2015-03-07:
Thanks- companion appears to prevent unsupported chars on other 'name' inputs, but not on the model name.

griebd, 2015-03-11:
Quite a simple solution, assuming the rules for model names are the same as for flight mode names....

in companion/src/modeledit/setup.cpp:
```
@@ -377,6 +377,8 @@ SetupPanel::SetupPanel(QWidget *parent, ModelData & model, GeneralSettings & gen
 
   ui->setupUi(this);
 
+  QRegExp rx(CHAR_FOR_NAMES_REGEX);
+  ui->name->setValidator(new QRegExpValidator(rx, this));
   ui->name->setMaxLength(IS_TARANIS(board) ? 12 : 10);
 
   if (firmware->getCapability(ModelImage)) {
```

though I use git, I'm new to GitHub.... tried to assign the issue, but I probably need permission for that, right?

projectkk2glider, 2015-03-11:
>though I use git, I'm new to GitHub.... tried to assign the issue, but I probably need permission for that, right?

Correct, you need to be a member of OpenTX github organization. But before you become a member you can fork OpenTX repository, make changes there and create a pull request against original OpenTX branch. You should base your work on OpenTX `master` branch which contains 2.0.x development.

commit 51a8da25 (2015-03-13): Fix #2122 on next branch
commit 9b074931 (2015-03-13): Merge pull request #2142 from griebd/limit_special_characters_in_model_name_2122

Fix #2122 on next branch
closed 2015-03-13 by bsongis