opentx/opentx #2124: Companion 2.0.15 issue

Ray58, 2015-03-07:
The play track fonction used on Taranis (any track chosen) using L1 to L32 logical switches does not seem to work when using simulation in Companion 2.0.15. The same command works on the Taranis itself.

Ray

kilrah, 2015-03-07:
The simulator does not yet have sound playback capability, next version will.

closed 2015-03-07 by kilrah

Ray58, 2015-03-07:
Thank you.

Ray

<div>-------- Original message --------</div><div>From: Andre Bernet <notifications@github.com> </div><div>Date:03-06-2015  11:13 PM  (GMT-05:00) </div><div>To: opentx/opentx <opentx@noreply.github.com> </div><div>Cc: Ray58 <raymond.boridy@gmail.com> </div><div>Subject: Re: [opentx] Companion 2.0.15 issue (#2124) </div><div>
</div>The simulator does not yet have sound playback capability, next version will.