opentx/opentx #2125: Mixer Monitor and Channel Monitor are different at gauge  all is 100%

hrenz, 2015-03-07:
Test next from today 07.03.2015
Mixer monitor and channel Monitor are different at gauge
Input 100%,  Mixer 100%, Servo 100%
[image]
[image]

davxlw, 2015-03-07:
It's not a bug but a feature ;)
The mixers gauges can show a larger range of value (200%) than the channels ones.
This view is better used with the channels values expressed in %, it allows to see if your mixes exceed your servos ranges.

closed 2015-03-07 by kilrah

bsongis, 2015-03-07:
This should not be the default though, is it?

bsongis, 2015-03-07:
Ah titles sind auf Deutsch geschrieben, ok!

hrenz, 2015-03-07:
yes self compiled, makefile with mixermonitor YES
there is no info that mixer Monitor gauge is 200% , because mixer and channel value is standard