opentx/opentx #2127: No Global Variables Menu : opentx-9xrpro-2.0

panguitou, 2015-03-08:
Despite selecting the gvars option I do not get the Global Variables Menu.

[image]

I tried many configurations but I never get the gvars menu, for the above exemple I have obtained 
only 11 menus 

[MODELES][CONF.MODELES][PHASES DE VOL][MANCHES][MIXEUR][SORTIES][COURBES][INTER LOG.][FONCTIONS SPEC.][TELEMESURE][GABARITS]

Shall I try the english version ?

Thanks in advance !
JM

projectkk2glider, 2015-03-09:
Gvars are at the bottom of Curves screen.

closed 2015-03-09 by bsongis

panguitou, 2015-03-09:
Thank so much for your very prompt reply...
But sorry to disagree, thanks to you I dig a little more and I finally found them not in the curves screen, but logically in each Flight Mode screen !
Regards

projectkk2glider, 2015-03-09:
True, GVARS are:
* on Flight mode screen in Companion
* on Curves screen (9X) or Flight mode screen(Taranis) on the radio