opentx/opentx #2128: [Companion 2.0.15] Issue with SD Structure Path

LapinFou, 2015-03-08:
I'm not able anymore to simulate LUA script. After some search it appears that if there is a "foreign" character in the path, then Companion will fail to find the SD card.
Example, if I use this path, the I got the issue:
C:\Users\Sébastien\Documents\Taranis\SDCard\Seb
When Companion failed to open the specified path, it will use this default path:
C:\Program Files (x86)\OpenTX 

If I'm using this kind of path (no French character), then it is working fine:
D:\Taranis\SDCard\Seb

LapinFou, 2015-04-16:
Anyone has been able to reproduce this problem?

projectkk2glider, 2015-04-16:
Press F6 while in simulator window to open the debug console and look there for the path that Companion is trying to open.

LapinFou, 2015-04-16:
Thanks. I did used F6. I discovered that Companion 2.0.15 wasn't handling anymore foreign character.
C:\Users\S<b>é</b>bastien\...  The "<b>é</b>" in the path is the issue.
However it was working well with previous Companion versions. My current workaround is to move my "Taranis" folder in another place with no "weird" character.
However, I opened a ticket to inform you that something was broken.
:)

jtaylor2, 2015-04-16:
I can confirm the problem on Windows 7 using downloaded Companion 2.0.15 and self compiled master (compiled on Windows 7 with cygwin).

The problem does not exist on next compiled with cygwin on Windows 7, but I haven't been able to figure out what fixes it on next.

I have a us/en system so am testing my adding a è (alt code 0232) to my sd path. C:\programs\OpenTX\SDèCard.  On master the console shows a ? for the è character and can't find the path and on next it displays the è correctly.and finds the paths.  I'm still looking at this, but:
git diff origin/master origin/next | grep -i sdpath in companion/src doesn't turn up anything obvious.

Jim

jtaylor2, 2015-04-17:
The problem is that simuSdDirectory[i]=path.at(i).toAscii(); in companion\src\firmwares\opentx\opentxTaranisSimulator.cpp does not translate foreign characters correctly:

This code in opentxTaranisSimulator.cpp produces the console output below it:

```
OpentxTaranisSimulator::OpentxTaranisSimulator()
{
  taranisSimulatorBoard = GetEepromInterface()->getBoard();
  QString path=g.profile[g.id()].sdPath()+"/";
qDebug() << "\nPath from g.profile ";
qDebug() << path;
  int i=0;
  for (i=0; i< std::min(path.length(),1022); i++) {
    simuSdDirectory[i]=path.at(i).toAscii();
  }
  simuSdDirectory[i]=0;
qDebug() << "\nPath from simSdDirectory";
qDebug() << simuSdDirectory;
}
```

```
trying OpenTX for FrSky Taranis import... version 216 variant 0 ok

Path from g.profile  
"/cygdrive/c/programs/OpenTX/SDèCard/" 

Path from simSdDirectory 
/cygdrive/c/programs/OpenTX/SD�Card/ 

Path from g.profile  
"/cygdrive/c/programs/OpenTX/SDèCard/" 

Path from simSdDirectory 
/cygdrive/c/programs/OpenTX/SD�Card/ 
f_opendir(/cygdrive/c/programs/OpenTX/SD�Card//SOUNDS/en/SYSTEM) = error 2 (No such file or directory)
f_opendir(/cygdrive/c/programs/OpenTX/SD�Card//SOUNDS/en/SYSTEM) = error 2 (No such file or directory)
f_opendir(/cygdrive/c/programs/OpenTX/SD�Card//SOUNDS/en/Extra_300s) = error 2 (No such file or directory)
findTrueFileName(/cygdrive/c/programs/OpenTX/SD�Card//BMP/Extra300-1.bmp)
        not found
f_open(/cygdrive/c/programs/OpenTX/SD�Card//BMP/Extra300-1.bmp) = INVALID_NAME

```

This code stream has been changed and the problem doesn't exist on next, so I guess the question is do we fix it on master or leave it go to 2.1.

jtaylor2, 2015-04-17:
Looking at this a little more I think it should be fixed in 2.0.16 because it will also prevent sound from working in Companion for anybody that has a foreign character in their SDcard path.  Sky9x uses the same code so it will have the same problem.

Unless somebody thinks it shouldn't be fixed on master I'll back port the logic from next and  put together a pull request.

projectkk2glider, 2015-04-17:
I think it should be fixed. Do it :+1:

closed 2015-04-17 by projectkk2glider