opentx/opentx #2129: Raw input marking

mhotar, 2015-03-08:
Hello,
I feel a little lack of graphics to distinguish the physical inputs/formatted inputs.
At least raw gimbal inputs need some icon :) 

[image]

bsongis, 2015-03-09:
Really nice!

closed 2015-03-09 by bsongis

mhotar, 2015-03-10:
Perfect ;)

Let's continue then :)  The timer icon, and sound/voice icons.

I am not sure if they are to be used globally or to use them in the custom language font only, for use in the languages where it is not easy to find short enough expression for particular sentence.

[image]
[image]

[image]

kilrah, 2015-03-10:
Don't agree with those. 
The graphics IMO are nice when they are there to complement info you already have (like with the sources), not when they replace the data altogether as they are often confusing on their own (just like the graphical menu discussion).

mhotar, 2015-03-10:
Ooops, Kilrah is back. So am I at least successful in 50% here?

I agree here, It is better to use text, when you can translate it to fit to given space meaningfully.

bsongis, 2015-03-10:
I prefer to remain as it is now. The sticks oees are especially good to differenciate from inputs with the same name. And I like a lot the trim one, I was happy to remove the "tRl" "tRr" "tEd" "tEu" "tTd" "tTu" "tAl" "tAr"

mhotar, 2015-03-11:
You both convinced me again :)

dkisselev, 2015-03-28:
I was working on another change with custom functions and found that these changes broke how trims are rendered on the 9XR:

[image]

Just wondering what the best way to fix it would be, a conditional define that renders the old tRl style labels, or is it possible to port the icon font over?

commit 5c980955 (2015-03-09): Fixes #2129 - Only EN and FR done, I will leave the others to the
translators