opentx/opentx #2135: add auto-save feature to companion

supermax49, 2015-03-11:
As an enhancement please consider adding an auto-save feature to companion so that .eepe files being edited are periodically saved.

mpjf01, 2015-03-11:
If added, should be optional and not the default. I don't want to find my working configuration accidentally overridden by a current experimental development.

kilrah, 2015-03-12:
Yes. Autosave is actually the last thing I want as it's guaranteed to save at the wrong moment with half-done things.
Work on something and save when you've finished it. Not that hard, and it's a reflex you need to use in anything IT anyway.

projectkk2glider, 2015-03-12:
This might only make sense if auto-save feature would use some kind of auto-numbering scheme for auto saved files. This way true file won't be overwritten by auto-save feature. But it complicates things and it is questionable how many users will actually benefit from this and then will be able to find the auto-save files if they need them.

romoloman, 2015-04-27:
I fully agree with kilrah, only it could be meaningful to have an auto savestate with autorecovery at next startup in the case companion crashes or the PC is turned off, but in several years I'm using companion it never crashed. So I think that the efforts to implement this feature are not justified and probably more than the human resources available.

closed 2019-11-15 by stale[bot]