opentx/opentx #2137: 9XR-PRO - wav files of more than 6 letters does not appear in the pick list

panguitou, 2015-03-12:
Firmware opentx-9xrpro-2.0
The wav files with a name of more than 6 letters does not appear in the play track picklist and cannot be select : 
[image]

On the opposite all the files even with longer name are accessible in the SD-HC Card screen :
[image]

To overcome this problem the wav files have to be renamed with maximum 6 letters name.

Thanks in advance
JM

kilrah, 2015-03-12:
It's normal, 6 chars is the max length that can be stored for names in special functions on sky9x/9XR-Pro.

closed 2015-03-12 by kilrah

panguitou, 2015-03-13:
Merci, désolé pour le dérangement.
JM