opentx/opentx #2140: Taranis speaks !!

lepicfoto, 2015-03-13:
Hi all, a question?
Sounds !
I set mine to play one time and not during startup
the taranis does not care :-(
Maybe a little bug in the program ?
It does not matter what I do it works only once, the first time I turn on the radio after I uset companion, all subsequent times, the taranis is talking to me at startup, it's not a problem just annoying, Taranis should not speak during startup
other than sticks pots "warnings" ?
Btw, and I'm on 2,0,15 everywhere.
all the best greetings from denmark.

deaconFlyer, 2015-03-17:
I had a similar problem. I replaced the MicroSD card and the problems went away. 

There have been numerous issues with sound .... flakey cards was one of them. 

Right now I am running a Class 10, which is probably overkill, but I had one lying around so why not.

kilrah, 2015-03-17:
Would help to know which sounds are playing and shouldn't, as well as seeing your SF configuration.

lepicfoto, 2015-03-17:
Hallo kilrah and all.

It's no harm, it's just a bit annoying and a little unprofessional,
especially when ther is crowd of spectators :-)

The original SD cards are set aside for backup,
and it is a brand new 2Gb SD card which is a clone of the original
in the Taranis.

The sounds in use are:(Full Amber soundpack)

engoff,
engon,
flapsup,
flapsdn,
airbk,
deplyd,

The sounds that are playing at start up and shouldn't:

engoff,
flapsup,

SD Card is setup like this:
SD Card name "TARANIS SDC"

/BMP/Splash Screens/ (a bunch of BMP files)

/EEPROMS/Backup/ (a bunch of EEPE files)

/FIRMWARES/FLVSS Smart Port LiPo Voltage Sensor/ (From frsky web site)
/FIRMWARES/Taranis X9D/ (From frsky web site)
/FIRMWARES/Vario - High Precision/ (From frsky web site)
/FIRMWARES/X8R Receiver/ (From frsky web site)

/LOGS/ (Logfiles from my "Hvid" Cub j-3)

/MODELS/ (one handful BIN files)

/SCRIPTS/BMP/ (empty)
/SCRIPTS/FUNCTIONS/ (empty)
/SCRIPTS/GAMES/ (snake)
/SCRIPTS/Hvid/ (empty)
/SCRIPTS/MIXES/ (empty)

/SOUNDS/en/system/ (Fulll amber soundpack)

the best greetings from denmark
Jesper Malling

kilrah, 2015-03-17:
SD contents are OK, what I'd like to see is your special functions page, or even better just post your entire eepe :)

lepicfoto, 2015-03-17:
Hallo kilrah and all.

I  hope that it works this way, I  just renamed the file to hvid.JPG
If not then please help

the best greetings from denmark
Jesper Malling

[image]

LapinFou, 2015-03-17:
Your link doesn't work.
"The image cannot be displayed because it contains errors."

kilrah, 2015-03-17:
Just save as and rename to eepe. 
I tried it on my radio and sounds don't play at startup (you have an annoying warning on the throttle mix though).

Do you have direct switch sounds on the card (sounds\en\Hvid\SAup.wav etc)?

lepicfoto, 2015-03-17:
Hi Kilrah

They tend to be silent first time I boot from the companion, then there is sound when I boot taranis up
no I is not got to move sounds yet, it is a project I'm working on:-)
and excuse the annoying throttel warning,:-) I'm working on it and take one small step at a time,

kilrah, 2015-03-17:
Really don't see why it would play the sounds. Can you take a little video of powering the radio that shows the problem so we can see if there's something weird with other sounds or the display?

lepicfoto, 2015-03-18:
ok will do, stand by :-)

lepicfoto, 2015-03-19:
Hi Kilrah
and all :-)

Sorry for the wait, work, spend by far too much time:-)
Here is a link to a small video.
link is for dropbox:

https://www.dropbox.com/sh/oyn8p7osvxl2r47/AADA50F7WnuoOZjDLjqcICYRa?n=44234057

And it is my tranis fresh of the computer, it start up nicely,
BUT when I start up at the field, then it is wrong again, and she speaks a little, every time, BUT fresh of the computer, :-( annoying. :-(
And then she pips! pips constantly, altso annoying.
Is it mee doing it wrong, i hope not, as i am reading the manual, a wise man ons told me Reat the F¤#&%"/ manual, and then and only then ask. :-]

Bedst regards
Jesper

kilrah, 2015-03-19:
The beeps are as I said due to the "warning" setting on your throttle mix.
I really can't reproduce the sounds issue.

What options are you using for the firmware?
Can you post both the EEPROM.BIN and FIRMWARE.BIN files from the virtual USB drive to your dropbox so I have the exact same config?

lepicfoto, 2015-03-20:
Hello Kilrah.
And all.

Thank to kilrah for pointing me in the right direction,
it was in the MIX and NOT in the setup,
ther was a quite annoying peep sound, when I start my traranis,
it is now completely gone and that's the problem with the taranis "speaks" also, all gone.
what can I say, setup / MIX, NOT quite the same Hehehe

I can post bin files on my d-box if you'd like them:-)
seen from my side, my problems just disappeared hehehe nice TNX.
that is to say until new ones pop up, and I think they probably will,

Again TNX to Kilrah

Best regards
Jesper Malling

kilrah, 2015-03-20:
OK, no issue then ;)

closed 2015-03-20 by kilrah