opentx/opentx #2144: GPS position: fractional seconds lack leading zeroes

mox17, 2015-03-16:
When showing a GPS position, the longitude could display as: 25°E 39'18.6  where the correct display would be: 25°E 39'18.006

The snip below from view_telemetry.cpp appears to be code that displays the coordinates. A LEADING0 option and a length of 3 for the ss % 1000 part would fix this problem.
The full seconds could also be displayed to always be 2 characters wide with a leading 0.

```
void displayGpsCoord(uint8_t y, char direction, int16_t bp, int16_t ap)
{
  if (frskyData.hub.gpsFix >= 0) {
    if (!direction) direction = '-';
    lcd_outdezAtt(TELEM_2ND_COLUMN, y, bp / 100, LEFT); // ddd before '.'
    lcd_putc(lcdLastPos, y, '@');
    uint8_t mn = bp % 100; // TODO div_t
    if (g_eeGeneral.gpsFormat == 0) {
      lcd_putc(lcdLastPos+FWNUM, y, direction);
      lcd_outdezNAtt(lcdLastPos+FW+FW+1, y, mn, LEFT|LEADING0, 2); // mm before '.'
      lcd_vline(lcdLastPos, y, 2);
      uint16_t ss = ap * 6;
      lcd_outdezAtt(lcdLastPos+3, y, ss / 1000, LEFT); // ''
      lcd_plot(lcdLastPos, y+FH-2, 0); // small decimal point
      lcd_outdezAtt(lcdLastPos+2, y, ss % 1000, LEFT); // ''
      lcd_vline(lcdLastPos, y, 2);
      lcd_vline(lcdLastPos+2, y, 2);
    }
    else {
      lcd_outdezNAtt(lcdLastPos+FW, y, mn, LEFT|LEADING0, 2); // mm before '.'
      lcd_plot(lcdLastPos, y+FH-2, 0); // small decimal point
      lcd_outdezNAtt(lcdLastPos+2, y, ap, LEFT|UNSIGN|LEADING0, 4); // after '.'
      lcd_putc(lcdLastPos+1, y, direction);
    }
  }
  else {
    // no fix
    lcd_puts(TELEM_2ND_COLUMN, y, STR_VCSWFUNC+1/*----*/);
  }
}
```

bsongis, 2015-03-16:
Will you propose a fix?

bsongis, 2015-03-16:
I will wait for some test before porting to 'next'

mox17, 2015-03-17:
I tested this on version 2.0.15 on a Taranis Plus
Below is a patch with my change.

/Erling

Make seconds and fractional seconds fixed length with leading zeroes.

```
erling@ubuntu:~/src/opentx/opentx-2.0.15/radio/src/gui$ diff -u  view_telemetry.cpp.old view_telemetry.cpp
--- view_telemetry.cpp.old	2015-01-14 00:17:41.000000000 +0200
+++ view_telemetry.cpp	2015-03-17 18:54:48.015109638 +0200
@@ -118,9 +118,9 @@
       lcd_outdezNAtt(lcdLastPos+FW+FW+1, y, mn, LEFT|LEADING0, 2); // mm before '.'
       lcd_vline(lcdLastPos, y, 2);
       uint16_t ss = ap * 6;
-      lcd_outdezAtt(lcdLastPos+3, y, ss / 1000, LEFT); // ''
+      lcd_outdezNAtt(lcdLastPos+3, y, ss / 1000, LEFT|LEADING0, 2); // ''
       lcd_plot(lcdLastPos, y+FH-2, 0); // small decimal point
-      lcd_outdezAtt(lcdLastPos+2, y, ss % 1000, LEFT); // ''
+      lcd_outdezNAtt(lcdLastPos+2, y, ss % 1000, LEFT|LEADING0, 3); // ''
       lcd_vline(lcdLastPos, y, 2);
       lcd_vline(lcdLastPos+2, y, 2);
     }
```

bsongis, 2015-03-17:
It's exactly what I commited 24 hours ago, no?

mox17, 2015-03-17:
Sorry - I was not paying attention to that you committed. You asked over email if I would propose a fix, so I just got started with that, i.e. got a build environment established, modified, built and tested the code on my radio and did the same thing as you.
I think you commit looks just fine :-)

commit 80159836 (2015-03-16): Fixes #2144 - Needs tests
commit 25ce79f9 (2015-03-16): Fixes #2144 - Needs tests
commit 4e1f04fa (2015-03-18): Fixes #2144 - Needs tests
closed 2015-03-18 by bsongis