opentx/opentx #2147: Triggering SWR=255 and TX Antenna problem with model configuration

sengel, 2015-03-17:
### Setup
* FrSky Taranis Plus (bought in 11/2014) with XJT using the stock firmware (i.e. not the EU FrSky firmware 1.8.1)
* Moved switches around for SF and SH (no soldering necessary in Taranis Plus, just moved the switches)
* Behaviour reproducible with OpenTx 2.0.13 and 2.0.15
* Behaviour reproducible with stock switch setup, i.e. SF with toggle switch and SH with momentary switch
* Displaying SWR on telemetry page
* Logging data to SD card, every 0.2 secs
* SF is used during 'Launch mode'
* Tested on the bench, not while flying (luckily)
* Tested with receiver off (also reproducible with receiver on)

Might be related to or already fixed with #2015.

Currently I am learing how to configure the Taranis. I try to setup a Libelle DLG using my own configuration and playing around with the "F3K Setup to work with F3K Training Script Version:V3" from here http://rcsettings.com/index.php/viewdownload/11-sailplanes/200-f3k-setup-to-work-with-f3k-training-script. The latter one is a lot more complex. I use it to get to know what is possible and learn ways on how to achieve it.

### Problem
#### Using F3K Setup
When using the "F3K Setup" and pulling (and holding) SF, the value for SWR immediately goes up to 255 and stays there for a short time (sometimes for a second, sometimes for longer). Then the displayed value for SWR drops to 193, then 127, then 63 and at last to 0. While trying to reproduce the behaviour the values (193, 127, 63) are displayed within a range from +- 2. So it is obvious that these values are in steps of 64.

While holding SF the chances are high to trigger an "TX Antenna problem" while SWR has a high value. This is not always happening but quite often.

The CSV logs display something like this, when using SF:
```
Date,Time,SWR,RSSI, ... ,SE,SF,SG,SH
2015-03-17,20:27:53.020,0,0, ... ,-1,-1,-1,-1
2015-03-17,20:27:53.220,0,0, ... ,-1,-1,-1,-1

2015-03-17,20:27:53.420,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:53.630,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:53.840,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:54.050,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:54.250,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:54.460,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:54.660,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:54.860,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:55.060,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:55.260,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:55.460,-,0, ... ,-1,1,-1,-1
2015-03-17,20:27:55.660,0,0, ... ,-1,1,-1,-1
2015-03-17,20:27:55.860,0,0, ... ,-1,1,-1,-1
2015-03-17,20:27:56.060,0,0, ... ,-1,1,-1,-1
```

'-' seems to be written if SWR has a value of 255 or simply is out of range. This happens at the beginning of pulling SF (-1 -> 1). A short time later (while still holding SF), SWR is show as 0. The intermediate values shown on the display (193, 127, 63) are not logged.

### Using my own setup
When using my own setup, I could not reproduce this behaviour. My setup is way simpler, just a little bit more than a basic configuration. Also, the CSV logs here are clean. Whereas in the above excerpt '-' is logged for SWR if SF is pulled, in my setup SWR is then still logged with 0.

### Remarks
At first I thought this was a hardware problem as I moved the switches just before I encountered this behaviour or a problem with the recently flashed OpenTx 2.0.15. But after I reflashed 2.0.13 and moved the switches back to their old positions, I found the behaviour with the model configuration.

I think the more complex "F3K Setup" somehow triggers the SWR bug and finally the antenna problem. Maybe there is too much going on whith this config during "launch" (SF).

I saw that #2015 describes this kind of behaviour with relation to telemetry. As I am still learning how to configure the TX, I am not quite sure, what the "launch mode" all does in the "F3K Setup" config. So I am not sure, if this is a duplicate of #2015 or a new bug.

bsongis, 2015-03-18:
Did you try the nightly builds?
http://jenkins.open-tx.org/nightly-20/

sengel, 2015-03-18:
Not yet. I will give the latest one a try.

I am using Linux. Do I have to use the corresponding Companion from the nightly build directory? Or is the official version 2.0.15 sufficient?

projectkk2glider, 2015-03-18:
>Do I have to use the corresponding Companion from the nightly build directory? Or is the official version 2.0.15 sufficient?

You can use 2.0.15.

sengel, 2015-03-18:
I just tested with the current nightly build version (2015-03-18 21:54:01) for Taranis Plus. Again with no receiver.

I tried multiple times with both model configs. The good thing is, that I could not reproduce the behaviour. SWR did not go up to 255 as I pulled and held SF for a longer time. And as such I also could not provoke the antenna problem.

But I encountered a another difference with SWR in between the two model configurations.

When just pulling and holding SF, the value for SWR stayed at 0 with both configs.

Then I put my hand around the antenna and the SWR readings went up to about 15 to 20. That's what I expected. Now I pulled SF, my hand still around the antenna.
* When using my simple config, SWR was always displayed with about 15 to 20 and also logged thisd way in CSV. When removing my hand while still holding SF, SWR also went back to 0. As expected.
* Now the same with "F3K Setup". Hand around antenna and SWR went up to about 15 to 20. Now pulling SF and SWR immediately dropped to 0 for a short time (about 1 second) and then went up to the former SWR value again. Taking a look into the log file shows the same: SWR is logged with a value of 17 and dropped to 0 when pulling SF. 1.2 seconds later SWR is logged with 17 again until I removed my hand from the antenna.

So something with the "F3K Setup" triggers this odd behaviour. Maybe it's too complex and uses too many cpu resources?

sengel, 2015-03-18:
Another update. I thought this behaviour had something to do with the SF switch, because the connection of SF is routed very near the antenna on the PCB. But just ignore the switches.

I reprogrammed the 'launch' switch from SF to SH and the behaviour is the same as described in my last comment (SWR value of 17 with drop to 0 when 'launching' and back to 17 while holding my hand around the antenna).

It seems that the launch mode of the "F3K Setup" model config is the cause of this behaviour.

closed 2019-11-15 by stale[bot]