opentx/opentx #2148: voltage telemetry should change to zero if sensor disconnect from battery

offer-shmuely, 2015-03-18:
Voltage Telemetry does not reset if sensor disconnect from battery

Setup:
OpenTx 2.0.12
Voltage Sensor: FRSKY SP-FLVS - SMART PORT LIPO VOLTAGE SENSOR
Revicer x8R

Expected Behavior:
   In case voltage sensor disconnect from battery, telemetry show 0 volt

Actual behavior:  
   In case voltage sensor disconnect from battery, telemetry show the last voltage

Problem:
   Since one (I am) may use the telemetry of to decide when to land, 
   the low voltage alarm will never operate, and the plane may crash.

kilrah, 2015-03-18:
As far as I know it's impossible for us to know if the sensor got disconnected form the battery with the data we get from the telemetry.

offer-shmuely, 2015-03-18:
Does the sensor itself send the same values forever?

How can it be verified if it the sensor or the OpenTX?

kilrah, 2015-03-18:
I just checked in the 2.1 branch and the sensor does stop sending values if disconnected. So that feature will be supported (and is already implemented) in 2.1, every telemetry value will have a timeout flag.

closed 2015-03-18 by kilrah

offer-shmuely, 2015-03-18:
Thanks!
Is there a milestone for 2.1?

kilrah, 2015-03-18:
If you mean estimated release time frame... No.

bsongis, 2015-03-18:
No timeframe, but it's already available in nightly builds