opentx/opentx #2150: OpenTx for 9X: GVAR minor bug/enhancement

vian31, 2015-03-18:
Hi all, I am really impressed by the Opentx software, thanks for that great job.

I would like to report a minor bug in Campanion about the GVAR fields on the FM page: the software is not recording the names of the GVARs when you exit the soft.
At least, as an improvement, it would be very appreciated to have the GVARs' names on the OpenTX GVARs' page (IE the CURVES page).
On the other hand, it would be very usefull to have more than 5 GVARs in OpenTX for 9X devices.
Some complex programs could use 2 or three more.

Thanks !

kilrah, 2015-03-18:
GVAR names are not available on the 9x, not enough resources.
Same for the 5 GVAR limit, you won't get more on a 9x...

kilrah, 2015-03-18:
Fixed... not the way you'd like unfortunately :(

closed 2015-03-18 by kilrah

vian31, 2015-03-19:
Thanks for this clear answer kilrah, neverthless could you please fix the recording bug on Companion ? It should be very usefull to have the names of GVARs in the computer's soft as a reminder.
Rgds

kilrah, 2015-03-19:
It's not a bug, companion does not have a way of recording anything separately from the radio's eeprom.

The bug was that GVAR names shouldn't have been displayed in companion for the 9x, and that's what I corrected.

vian31, 2015-03-19:
OK thanks.
Have a nice day.
Vianney

commit ce2a2788 (2015-03-18): GVAR names not available for 9x in companion (Fix #2150)
commit 8d590904 (2015-03-18): GVAR names not available for 9x in companion (Fix #2150)