opentx/opentx #2154: File system corruption with fullish large SD card

jmrware, 2015-03-20:
Summary
-------

When a known-good largish SD card (16GB) more than half full is plugged into the Taranis (X9D or X9DP running OTX 2.0.13+), and the Taranis is plugged into a PC, (from bootloader mode), CHKDSK reports many errors on that removable drive (and attempts to read certain files from the drive fail). But take this same SD card and plug it back into the PC using a standard SD card adapter (not the Taranis), then CHKDSK reports zero errors (and all files can be accessed without errors). Note that because of this bug, the files used to fill up the disk must be copied using the external card adapter - attempts to do so when connected via the Taranis won't work.

Background
----------

I'm getting ready to release "TaraniJuke", which is a collection of some LUA scripts which implement a rather useful background music player. It works just fine with a 2GB card, but I started noticing some problems when I moved up to a larger SD card (16GB). Once the SD card starts filling up, the Taranis starts experiencing problems accessing the SD card. These problems accessing the card only appear when the SD card is connected to the computer using the Taranis - these problems do not occur when the SD card is connected using a standard SD card adapter. It is unknown if this error manifests under normal operating conditions when the Taranis is not plugged into a computer.

How to reproduce
----------------

Download the following ZIP file:

ftp://ftp.xmission.com/pub/users/j/jmr/outgoing/TARANI_FS_BUG_TEST.zip

MD5: D3D4884F0CA7A1DFF1EC68EA8C42D7C3
SHA-1: E9BE01D7EA6228AE6D8173B8086E8CF988744F1B

This zip archive contains 3 files:
DUPWAV.BAT - Batch file to make 1000 copies of WAV file in the SOUNDS\en folder.
README.TXT - Description of how to reproduce the error.
TEST.WAV - A 32-kHz, mono Taranis compatible WAV file that is roughly 10MB in size.

Here are the contents of the README.TXT file from the archive:

README.TXT
----------

How to reproduce the Taranis FatFs bug:

1. Place a 16GB MicroSD card in an external USB adapter and plug it into a Windows PC. It 
will show up as a removable drive "X:" in Windows Explorer (however, the drive letter on 
your system will likely be different from X). Do NOT use the Taranis as the USB adapter at 
this point.
2. Open Windows Explorer and right click on the removable drive X:. Format the SD card as 
FAT32 using the default settings (cluster size = 8KB). Do NOT select the "quick format" 
option.
3. Copy the standard OpenTX files and directories to the SD card. It should look something 
like the following:
    Drive X: Directory tree:  
        +---BMP  
        +---EEPROMS  
        +---FIRMWARES  
        +---LOGS  
        +---MODELS  
        +---SCRIPTS  
        |   +---MODEL01  
        |   +---FUNCTIONS  
        |   +---MIXES  
        |   \---WIZARD  
        \---SOUNDS  
            \---en  
                \---system  
    (The only mandatory folder to be copied is: SOUNDS\en)  
4. Copy DUPWAV.BAT and TEST.WAV (from this zip archive) to the root directory of the SD 
card, i.e. "X:\".
5. From Windows Explorer, run X:\DUPWAV.BAT by double clicking on it. This batch file simply 
makes 1000 copies of the TEST.WAV file (which is 12MB) to the SOUNDS\en folder giving each 
copy of the file a unique name: _TST0001.WAV, _TST0002.WAV, ... _TST1000.WAV. Note that this 
process takes a LONG TIME ~1 hour. (Go take a break!)
6. Once the batch file finishes making all the wav file copies, verify that the file system 
is A-Ok; Double click on X:\SOUNDS\en\_TST1000.WAV and verify that it plays OK. Open a DOS 
Command Prompt Window and type: "CHKDSK X:" (Where "X" is the drive letter for the SD card.) 
Verify that CHKDSK reports no errors.
7. Dis-mount the SD card by clicking on the "Safely Remove HArdware" icon in the system tray 
(for WIn XP -  On Win7 and higher, right click the removable drive X; in Explorer and 
select: "Eject"). Remove the MicroSD card and insert it into the Taranis SD card slot.
9. Connect the Taranis to the PC by using the 3-finger-salute bootloader method. The SD card 
in the Taranis shows up as removable drive "Y:"  (But once again, the drive letter on your 
system will likely be different from Y:). At this point, things should start going wonky.
10. Repeat the tests described in step 6 to test the file system integrity. At this point, 
Windows should experience a variety of problems accessing the drive - it will be unable to 
play the later WAV files and will report many errors when running CHKDSK from the command 
line. Do NOT modify any of the files on the SD drive.
11. Dismount the Taranis, remove the SD card and remount it using the external adapter. The 
card should once again be A-Ok.

kilrah, 2015-03-20:
> Copy DUPWAV.BAT and TEST.WAV (from this zip archive) to the root directory of the SD card, i.e. "X:\".
From Windows Explorer, run X:\DUPWAV.BAT by double clicking on it. This batch file simply makes 1000 copies of the TEST.WAV file (which is 12MB) to the SOUNDS\en folder giving each copy of the file a unique name: _TST0001.WAV, _TST0002.WAV, ... _TST1000.WAV. Note that this process takes a LONG TIME ~1 hour. (Go take a break!)

Run the bat in a local directory on the PC and then copy the result - will be loads faster ;)

Will have a look.

jmrware, 2015-03-21:
Get the files onto the SD card any way you please. The important thing here is that The Taranis  experiences read and write errors once a certain threshold number of (allocation units/directory entries/total sectors/some-other-metric) are exceeded. Note that this problem occurs even when the SD card is formatted with a large cluster size (64K) as well as with the default (8K).

I have no problem with the Taranis being slow or unable to complete a file IO operation. But it should fail gracefully if it does not have the resources needed to complete an operation. It clearly should NOT go ahead and corrupt files and directories, (and this is what my testing shows to happen in a repeatable manner).

mpjf01, 2015-03-21:
From what has been said it seems that the problem does not happen on the 2GB card supplied with the radio. If so, why not just stick with that?

kilrah, 2015-03-21:
I can confirm the problem when accessing through USB, don't get a chkdsk prompt but just listing the big folder already shows there's an access problem.

[image]

But probably more importantly it works perfectly fine on the radio itself, so the problem would be in the USB mass storage access code, FatFS works fine.

fraternl, 2015-04-27:
I recently reformatted my 16GB SD-Card. For a few months I didn't get any SD-Card errors, but they re-appeared. Unfortunately 2 times when I was logging my first 2 thermal flights.
I also had 2 entries of a "LOGS" folder in my root.
I copied the contents to my harddisk and using an SD-CARD reader copied these back without the LOG-files.

Do you think it's related?

kilrah, 2015-04-27:
I wouldn't think so, this issue is quite clearly an usb access problem as the directories/files that (consistently) appear corrupted on the PC work perfectly fine on the radio itself.

fraternl, 2015-04-27:
But sometimes I copy files to the SD-Card using the USB-cable with the Taranis in bootloader mode. Couldn't the corruption start there?
It sure feels good that all this has no impact on the controls.
I wish I could be of help.

projectkk2glider, 2015-04-27:
I think I found the issue. 

The problem (for read) is in function `SCSI_Read10()` https://github.com/opentx/opentx/blob/2817866ad239f445eff9ec8ddc6c7197ef05703d/radio/src/thirdparty/STM32_USB-Host-Device_Lib_V2.1.0/Libraries/STM32_USB_Device_Library/Class/msc/src/usbd_msc_scsi.c#L475

Which first reads logical block address (LBA) from USB packet into uint32_t. This is ok:
```c
    SCSI_blk_addr = (params[2] << 24) | \
      (params[3] << 16) | \
        (params[4] <<  8) | \
          params[5];
```

It later does a stupid thing, it multiplies this number with the block size (which is 512 in our case):
```c
    MSC_BOT_State = BOT_DATA_IN;
    SCSI_blk_addr *= SCSI_blk_size;
    SCSI_blk_len  *= SCSI_blk_size;
```

This corrupts block adress for any addres bigger than 0x00FFFFFF. Later SCSI_blk_addr is divided again with block size, before beeing passed into OpenTX diskio driver.

Same problem in `SCSI_Write10()` function.

According to above the problem should appear any time we try to read/write data over 4G boundary.

projectkk2glider, 2015-04-27:
Found this patch https://gist.github.com/voidlizard/2202952

projectkk2glider, 2015-04-27:
Also found this commit that mentions this bug, but does not seem to address it https://code.google.com/p/opentx/source/browse/trunk/src/targets/taranis/STM32_USB-Host-Device_Lib_V2.1.0/Libraries/STM32_USB_Device_Library/Class/msc/src/usbd_msc_scsi.c?spec=svn2366&r=2366  

@bsongis you did this commit. Any clue?

bsongis, 2015-04-27:
Really interesting, thanks @projectkk2glider for your research!

projectkk2glider, 2015-04-28:
Bootloader after the patch:
```
   text	   data	    bss	    dec	    hex	filename
  32540	    292	  19684	  52516	   cd24	bootloader_ramBoot.elf
```

Applying above patch pushes the Bootloader size over 32kB! Not good. Maybe I can come up with a solution that does not use 64 bit variables.

fraternl, 2015-04-28:
Yay :-D :-D  +1: 
Many thanks for fixing this....

projectkk2glider, 2015-04-28:
@fraternl can you test it?

jmrware, 2015-04-28:
Is has been said that this bug only occurs when accessing the SD card while in bootloader mode.

Are all of the devs absolutely certain that this problem never occurs during normal operation?

projectkk2glider, 2015-04-28:
@jmrware We are pretty sure, but the proof will be in the testing. Are you willing to do a test? I can compile you a firmware image for you to use for testing. It should not mess up your models as you will only be changing the Bootloader.

closed 2015-04-28 by bsongis

jmrware, 2015-04-28:
@projectkk2glider - Sure I'd be willing to test it out, but it looks like testing is now a moot point as this bug issue is now closed.

p.s. Thanks for fixing this and for all your hard work guys.

projectkk2glider, 2015-04-28:
@jmrware testing is needed! Look at https://github.com/opentx/opentx/pull/2237#issuecomment-97171594

projectkk2glider, 2015-04-28:
I have now done another test with 8GB card using this fix and all apear to be working. The test procedure:
* put the SD card in USB reader and create three 2GB files, then copy to the card the standard contents of Taranis files (sounds, etc..) This makes sure that all interesting files are over 4GB mark.
* put the card in Taranis and start it. Sounds played normally, the SD Card browser worked, deleted one file for test.
* Turn on radio in Bootloader, connect to computer and compute md5sum of several sound files and compare it to original files. All the same. Then copy some PDF files to the radio. Eject.
* Put the card in card reader again and open PDF files. All were opened successfully. Run fsck on the card, reports no problems found.

jmrware, 2015-05-03:
I have tested the new 2.0.16 bootloader and can confirm that this issue appears to be fixed.

Thanks for the good investigative eye @projectkk2glider !

commit a1e467aa (2015-04-28): Fixes #2154: USB mass storage problem with cards bigger than 4GB
PR #2237 (2015-04-28): Fixes #2154: USB mass storage problem with cards bigger than 4GB
This attempts to fix large (>4GB) SD card problem without resorting to 64 bit variables and arithmetics.

I have tested this with my radio and 8GB SD card and looks like it is working. I insist that somebody else also tests this, before we merge into `master`!

Closes #2154
commit 1e504d04 (2015-04-28): Merge pull request #2237 from opentx/projectkk2glider/issue_2154_64bit_overflow

Fixes #2154: USB mass storage problem with cards bigger than 4GB
commit bd2699f2 (2015-04-28): Fixes #2154: USB mass storage problem with cards bigger than 4GB