opentx/opentx #2160: Companion 2.0.99 repeatedly accesses floppy disk.

Mow06, 2015-03-22:
Loaded Companion 2.0.99 on an WINXP SP3 desktop with a floppy disk drive. (Remember them?).

Floppy disk is accessed about once per second, repeatedely. No disk in drive, light flashes, drive clicks.
Open settings menu, close settings menu, floppy disk now accessed twice about every second.
Open a file and close, now three times every second.etc.

Tried another XP desktop and same condition.

Disabled floppy disk, don't use it anyway, but wondered if it indicates some other possible problem.

Doesn't happen on Companion 2.0.15. or previous versions used on this machine.

kilrah, 2015-03-22:
What are the different paths in the settings set to?

Mow06, 2015-03-22:
All set to C:/ drive settings, same as the 2.0.15 Companion settings. I've tried resetting everything and it still does it.

bsongis, 2015-03-23:
I think it's for the SD share button refresh, it polls all drives

projectkk2glider, 2015-03-23:
Two separate bugs.

First one is caused in `void MainWindow::updateMenus()` which calls `updateSdsyncAction();` Each call creates a new timer with 1s timeout. This explains the behaviour of 1, 2, 3 floppy acesses each second.

Second bug is access to floppy drive. It is caused by `findMassstoragePath()`which enumerates all drives and gets called periodically by `updateSdsyncAction();` Several possibilities to solve:
* we ignore drive letters A and B on Windows (that still leaves problem on Linux and Mac)
* we determine drive type for each device without access to device and skip it if it is floppy drive (not sure if this is possible, have to research....)

projectkk2glider, 2015-03-23:
First bug fixed, but the second one (skip floppy drives on Windows) is difficult. There is no simple solution, the best I have found is http://stackoverflow.com/questions/3156284/check-floppy-drive

I don't have a native Windows environment, so I am leaving this to someone else :)

fraternl, 2015-04-29:
If the problem is repeated access one could check the capacity and then leave out all with a capacity of less than a GB.

romoloman, 2015-05-01:
It would be possible to compile at runtime a blacklist of devices by using NtQueryVolumeInformationFile from winddk, at the first call of findmassstoragepath we compile the list by blacklisting all the devices that do not have the FILE_REMOVABLE_MEDIA flag set and that have FILE_FLOPPY_DISKETTE flag set.
In next call of the function we just skip drives in blacklist.
The only problem is that for calling NtQueryVolumeInformationFile we also need to install winddk.

bsongis, 2015-05-11:
Thanks to @romoloman I close this issue.

Mow06, 2015-06-27:
Been away for some time. Finally got to test this. 2015_06_27 is still accessing Floppy disk continuously.

fraternl, 2015-07-03:
First time I'm using Companion 2.1 on a machine with a Floppy drive.
It's still accessing it continuously on version 2015-07-03

kilrah, 2015-07-18:
Reopening that as there are still reports of the issue with release version.

bsongis, 2015-07-18:
I think I will make the button always enable, it will remove all problems!

closed 2015-07-18 by bsongis

kilrah, 2015-07-18:
Yeah, as long as an error is thrown if the button is pressed and the card isn't available it should be good!

projectkk2glider, 2015-07-18:
It is good, erorr looks like this:
[image]

PR #2244 (2015-05-05): Profilebackupfolder next
Fixes:
#2160 and #2080
commit 4b9f69a6 (2015-07-18): Fixes #2160
commit c0cc5351 (2015-07-18): Fixes #2160