opentx/opentx #2163: Show Now button (Request) for Companion Log Viewer

HuntTheShunt, 2015-03-23:
The next time someone is working on the Companion Log Viewer could a button be placed by the Reset button (or anywhere the developer thinks appropriate) the would activate the current selected available fields.

Currently as soon as you select an available field the program locks out the user because it is working in the background processing the users selection. 
The problem is the user has not completed his multi-selection yet and cannot do so because of the background processing.
The above spoils an excellent open-tx tool.

If the default status of the graph was to show nothing until the user pressed a button marked/entitled "Show" or a long one with "Display selected fields"

The above would make great tool easier to use.

Note: Apologies if this request has been placed incorrectly.

kilrah, 2015-03-23:
Hmm IMO the root issue is rather the poor performance.
Anytime I open a log and it's slow to respond to a selection I just close it and use Excel instead.

Allowing a multi-selection then clicking a button that then has you wait for 5 minutes is not really a solution. If it's too slow to process my multi-selection it's too slow to use at all.

projectkk2glider, 2015-03-23:
I agree about the slowness, it has to be fixed. I had a plan to look at this slowness issue, but haven't found the time yet.

griebd, 2015-03-23:
I am looking at this code, to fix #1863..., and the code is already running a bit faster, but I need a bigger test file (mines are short, my heli only flies like 5-8 min... ;). And I only make logs of it (my planes have no log) because the LiPo cells, so #1863 is a fix for me too actually.
Anyhow, I would tend to agree that we should make our selection and then click "plot"....
And I think we should limit the selection to what we have available, 2 y axis only. This way you wouldn't be able to select 'cells', 'temps' and 'rpm' at the same time, their ranges are too far different. But you could select all 12 'cells' at once and include all 'temps' in the same graph...

HuntTheShunt, 2015-03-23:
Let me have your email via PM and I will send you the longest I have.
Not sure how that is done in here. You can contact me by PM on rcgroups.com to TrueBuld

TrueBuild, 2015-03-23:
griebd ,
I can send you some log files if you contact me via email.

jtaylor2, 2015-03-23:
>And I think we should limit the selection to what we have available, 2 y axis only. This way you wouldn't be able to select 'cells', 'temps' and 'rpm' at the same time, their ranges are too far different. But you could select all 12 'cells' at once and include all 'temps' in the same graph, one y axis to 'cells' and the other to 'temps'...

I don't think limiting the selections to only two y axis is a good idea.  Sometimes you only want to see the relationship between the selections and displaying them the way it is now without y axis units is fine for that.  If the range is so different that it really isn't usable nothing says you have to do it again.

In general I think limiting what a user can do because you don't think it's useful, or that  they don't know what they are doing is a bad idea.  Particularity in a case like this where it would take away an already existing feature that people are already using.

Jim

griebd, 2015-03-23:
@jtaylo2, yes, I actually agree with you... :wink:
I guess I was mislead by laziness, it's a bit of a bigger work to let it enable after the changes I already did (and I actually couldn't see any use for that)... but o.k. them, I will get the log from @TrueBuild and keep working on logsDialog...

But what about the plot button? Should be implemented or not?

projectkk2glider, 2015-03-23:
I would prefer not having a *Plot* button. I like the plot that updates automatically. If we can not speed up the plotting, then maybe we can at least make a plotting interruptible - ie if user selects another item, abort current plotting and start a new one.

projectkk2glider, 2015-03-23:
I did a little profiling. The bulk of time is spent inside `QPainter::drawPolyline()` which is slow, one reference is https://bugreports.qt.io/browse/QTBUG-9208

The QCustomPlot that we use for graphing itself has some notes on performance. I just tried this on Linux box and difference is light years faster:
```
./companion -graphicssystem opengl
```

This means that we just need to calll `QApplication::setGraphicsSystem("opengl")` if OpenGL is avalilable on current running system. 

The appearance with OpenGL is a little different from stock, but acceptable I think.

projectkk2glider, 2015-03-23:
The culprit to slowness is the **antialiasing**.

projectkk2glider, 2015-03-23:
The profile which captures plotting first RSSI and then SWR values:
[image]

griebd, 2015-03-29:
Ok, did some test and the new function runs in close to half the time of the original one...
Did disable anti-alias on graphs, got a little improvement, but quite small.
Disabled a few options when loading the cvs file and got a significant improvement with only a few graphical differences...
Have a look at it and tell me what you think!

projectkk2glider, 2015-03-29:
Using my recommendation of 
```
./companion -graphicssystem opengl
```
produces extremely fast graphs, **but** it affects other parts of Companion, specially it negatively impacts the radio simulation:
* high CPU usage, channel output change is not smooth
* strange LCD display, half of pixels are missing

So it looks like OpenGL is not the right solution here.

griebd, 2015-03-30:
Well, worst in my opinion is that on some cases, like mine, you get lots of errors and no screen at all with opengl option... probably a lib missing or something, but that shows we will have lots of users complaining.... :wink:

TrueBuild, 2015-03-30:
Obviously better performance from the code would cure the problem; however, in the meantime the real problem is that program locks up because it is working hard plotting the moment the user selects one field.
A quick fix that would please most people would be a "plot now" button after a use has made his selection.
I do not mind waiting to see the graph.
The frustration comes from not being able to select fields.

Another option is to have another tab in the Edit Settings called View Log Settings with a default option set as now. The user could then select the option to have a plot now button with help text that says "Clicking this setting stops automatic plotting enabling to select multiple fields" or something similar that locks the plotting code till the user presses "plot now".

TrueBuild, 2015-03-30:
projectkkt2glider,
Would a shortcut "drive letter:\Taranis\Companion9x\companion.exe" -graphicssystem opengl work in windows 7?
I have 4.2 OpenGL

projectkk2glider, 2015-03-30:
>Would a shortcut "drive letter:\Taranis\Companion9x\companion.exe" -graphicssystem opengl work in windows 7?

I have no idea, try it :)

projectkk2glider, 2015-03-30:
I've also tried to turn of anti-aliasing for graph line(s). There is some improvement, but it still takes more than half the time of anti-aliased and looks horrible.

griebd, 2015-03-31:
Two points:
- I'm thinking in making things a bit smarter and draw graphs independently, as you choose or remove them. I believe its going to make a huge improvement in plotting time...
- BUT for some reason when you have some selections like a fly sessions or even selecting manually, the plotting time explodes... I guess it's something with "isSelected" but I haven't looked at it yet...

griebd, 2015-03-31:
Any reason not to update QCustomPlot? Had a look at it and it isn't such a big work.... Have no idea about performance differences though!!

griebd, 2015-04-14:
ok, got myself lost in the way.... kidding (partially)....
anyway, with the log files I have to test my huge complain now is when I'm opening the log files, will try to improve this later (and fix a small unnoticeable bug when you have a open file and open another)!

if you guys have the time, test the changes and see if you notice any improvement (or not).

Thanks!

TrueBuild, 2015-04-14:
I will test it if you have an exe or give me a tutorial on compiling the code on a windows platform.
An exe would give you faster feedback :)

kilrah, 2015-04-15:
The instructions for compiling are in the wiki :)

TrueBuild, 2015-04-15:
Kilrah,
Thanks for the hint; downloading and installing all the stuff now.

TrueBuild, 2015-04-15:
Having problems with SmartGit...
[image]
In the password box I am using my password for this site but that does not work.
What am I doing wrong?

TrueBuild, 2015-04-16:
Having trouble with path...
[image]
[image]
[image]
The path seems to XercesC looks ok?
Is my path too long?

kilrah, 2015-04-16:
Stuff that the guide says to put in C:\Programs\ MUST be in C:\Programs\. Several things are hardcoded to that at this point.

jtaylor2, 2015-04-16:
You didn't follow the instruction in the wiki which say:
 
Xerces
http://xerces.apache.org/mirrors.cgi#binary
You want V3.1.1 for Visual Studio 10
Unzip the contents of the zip to "C:\Programs\"
Rename the folder "xerces-c-3.1.1" to "xerces-vc" so you have "C:\Programs\xerces-vc\"

That's where the findxercesc.cmake file looks for it if you are using VC so you can either put it there or change your findxercesc.cmake to look for it where you have it.

```
C:\opentx\git\opentx\companion\src\cmake>pg findxercesc.cmake
# Find Xerces-C
# The following setings are defined
# XERCESC_ROOT_DIR, the root of the include and lib directory
# XERCESC_INCLUDE_DIR, the full path of the include dir (ADVANCED)
# XERCESC_LIBRARIES, the name of the xerces-c library (ADVANCED)

# Look for a root installation
IF( MSVC )
SET( XERCES_WINDIR C:/Programs/xerces-vc )
ELSE( )
SET( XERCES_WINDIR C:/Programs/xerces )
ENDIF( )

FIND_PATH(XERCESC_ROOT_DIR include/xercesc/parsers/SAXParser.hpp
  ${XERCES_WINDIR}
```
Jim

TrueBuild, 2015-04-16:
fair point.
C: is limited on space and G: is now effectively C: all very confusing.

However, since you say stuff is hard coded I will delete stuff on C: to make room and stick to conventions in future.

kilrah, 2015-04-17:
The "Programs" folder only takes 130MB...

Looks like you've got a huge mess of partitions with stuff spread over 3 of them...

TrueBuild, 2015-04-22:
Guys, I have created a clean XP (SP3) virtual pc and put everything where it should be.
I am having problems with the Xerces not being found again.
First build was with all as in the wiki (as far as I can see) and the build failed to find xerces.
So I tried a path to c:\program files\xerces-vc\bin and then c:\program files\xerces-vc\lib to no avail ;(

CMakeOutput.log...
The system is: Windows - 5.1 - x86
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler:  
Build flags: 
Id flags: 

The output was:
0
Microsoft (R) Build Engine Version 4.0.30319.1
[Microsoft .NET Framework, Version 4.0.30319.1]
Copyright (C) Microsoft Corporation 2007. All rights reserved.

Build started 22/04/2015 07:56:30.
Project "C:\OpenTxBuildDir\companion\build\CMakeFiles\3.2.2\CompilerIdC\CompilerIdC.vcxproj" on node 1 (default targets).
PrepareForBuild:
  Creating directory "Debug\".
InitializeBuildStatus:
  Creating "Debug\CompilerIdC.unsuccessfulbuild" because "AlwaysCreate" was specified.
ClCompile:
  C:\Program Files\Microsoft Visual Studio 10.0\VC\bin\CL.exe /c /nologo /W0 /WX- /Od /Oy- /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Fo"Debug\\" /Fd"Debug\vc100.pdb" /Gd /TC /analyze- /errorReport:queue CMakeCCompilerId.c
  CMakeCCompilerId.c
Link:
  C:\Program Files\Microsoft Visual Studio 10.0\VC\bin\link.exe /ERRORREPORT:QUEUE /OUT:".\CompilerIdC.exe" /INCREMENTAL:NO /NOLOGO kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /MANIFEST /ManifestFile:"Debug\CompilerIdC.exe.intermediate.manifest" /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /PDB:"C:\OpenTxBuildDir\companion\build\CMakeFiles\3.2.2\CompilerIdC\CompilerIdC.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:".\CompilerIdC.lib" /MACHINE:X86 Debug\CMakeCCompilerId.obj
  CompilerIdC.vcxproj -> C:\OpenTxBuildDir\companion\build\CMakeFiles\3.2.2\CompilerIdC\.\CompilerIdC.exe
Manifest:
  C:\Program Files\Microsoft SDKs\Windows\v7.0A\bin\mt.exe /nologo /verbose /outputresource:".\CompilerIdC.exe;#1" /manifest Debug\CompilerIdC.exe.intermediate.manifest
PostBuildEvent:
  for %%i in (cl.exe) do @echo CMAKE_C_COMPILER=%%~$PATH:i
  :VCEnd
  CMAKE_C_COMPILER=C:\Program Files\Microsoft Visual Studio 10.0\VC\bin\cl.exe
FinalizeBuildStatus:
  Deleting file "Debug\CompilerIdC.unsuccessfulbuild".
  Touching "Debug\CompilerIdC.lastbuildstate".
Done Building Project "C:\OpenTxBuildDir\companion\build\CMakeFiles\3.2.2\CompilerIdC\CompilerIdC.vcxproj" (default targets).

Build succeeded.
    0 Warning(s)
    0 Error(s)

Time Elapsed 00:00:05.92

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CompilerIdC.exe"

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CompilerIdC.vcxproj"

The C compiler identification is MSVC, found in "C:/OpenTxBuildDir/companion/build/CMakeFiles/3.2.2/CompilerIdC/CompilerIdC.exe"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler:  
Build flags: 
Id flags: 

The output was:
0
Microsoft (R) Build Engine Version 4.0.30319.1
[Microsoft .NET Framework, Version 4.0.30319.1]
Copyright (C) Microsoft Corporation 2007. All rights reserved.

Build started 22/04/2015 07:56:37.
Project "C:\OpenTxBuildDir\companion\build\CMakeFiles\3.2.2\CompilerIdCXX\CompilerIdCXX.vcxproj" on node 1 (default targets).
PrepareForBuild:
  Creating directory "Debug\".
InitializeBuildStatus:
  Creating "Debug\CompilerIdCXX.unsuccessfulbuild" because "AlwaysCreate" was specified.
[... 456 lines cut ...]
    Feature record: CXX_FEATURE:1cxx_long_long_type
    Feature record: CXX_FEATURE:0cxx_noexcept
    Feature record: CXX_FEATURE:0cxx_nonstatic_member_init
    Feature record: CXX_FEATURE:1cxx_nullptr
    Feature record: CXX_FEATURE:1cxx_override
    Feature record: CXX_FEATURE:0cxx_range_for
    Feature record: CXX_FEATURE:0cxx_raw_string_literals
    Feature record: CXX_FEATURE:0cxx_reference_qualified_functions
    Feature record: CXX_FEATURE:0cxx_return_type_deduction
    Feature record: CXX_FEATURE:1cxx_right_angle_brackets
    Feature record: CXX_FEATURE:1cxx_rvalue_references
    Feature record: CXX_FEATURE:0cxx_sizeof_member
    Feature record: CXX_FEATURE:1cxx_static_assert
    Feature record: CXX_FEATURE:0cxx_strong_enums
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:1cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:1cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates

[image]

jtaylor2, 2015-04-22:
Xerces should be under c:\Programs not "Program Files" unless you change findxercesc.cmake to point to where you have it.  You can't set a path to it, cmake needs to find it.  Put it under c:\Programs named like the wiki says and figure out why cmake can't find it.

griebd, 2015-04-22:
or just wait for the next nightly build, it's all merged already on next (2.1)...

TrueBuild, 2015-04-22:
Thanks jtaylor2.
It is very clear in the wiki that it should be programs.
I have not excuse :(
Changing it now.

jtaylor2, 2015-04-22:
@TrueBuild Any further build problems or questions should go to the forum as they are not really related to the issue.

griebd, 2015-04-25:
Looks like it's solved, right? If someone disagrees, feel free to reopen!

closed 2015-04-25 by griebd

projectkk2glider, 2015-04-25:
The log viewer speed is so much better now that any other changes are unnecessary. Great work @griebd :+1:

bsongis, 2015-04-25:
What about a 2.0 release beginning of next week?

projectkk2glider, 2015-04-25:
>What about a 2.0 release beginning of next week?

Yes I agree, there are no mayor issues for 2.0.16 open.

kilrah, 2015-04-26:
Yes I agree, there are lot of little quirks fixed that people will enjoy. Can always do a .17 later for outstanding things, but we need a release.

bsongis, 2015-04-26:
Ok. Today then?

romoloman, 2015-04-26:
There is still some minor issue (printdialog and comparedialog) that I'm fixing, but nothing blocking.

kilrah, 2015-04-26:
One thing that would be good to include in this release is an update / separation of the stamp locations for the major branches. As I've separated settings and install folders it would make sense to finish that part of the work... but I don't seem to have access to the server anymore.

LapinFou, 2015-05-05:
Sorry to post a message in a closed ticket, however it is the best way I found to thanks the work of griebd.
Using the log viewer is so smooth now!!! :+1: Thanks a lot for your work (and, off course, thanks to all the devs).

TrueBuild, 2015-05-05:
I would also like to thank Adriano Grieb for the stunning improvement in the Log File Viewer. :+1:

griebd, 2015-05-05:
thanks guys, just trying to give something back to this wonderful piece of work we have!!!!
flying is a lot nicer with OpenTX (and in my case, Taranis)