opentx/opentx #2172: GPS Distance goes 0 when traveling more than 4km

SamuelBrucksch, 2015-03-28:
After around 4200m the distance goes back to 0 and it starts counting up again.

Opentx version 2.0.13

kilrah, 2015-03-28:
How do you manage to fly more than 4.2km away from you with FrSky RF??

SamuelBrucksch, 2015-03-28:
I testet it with a gps emulator, because i want to use the telemetry for my antenna tracker project. I myself only flew ~4km, a friend of me already flew 6,5km and still had telemetry with the 60cm antennas fixed under the wings.

bsongis, 2015-04-02:
The maths we do here are good for small distances, I think we will keep them as they are on 9x. Only change on ARM boards as it will use more CPU

kilrah, 2015-04-02:
Yep by the way good question, what radio are you using?

SamuelBrucksch, 2015-04-02:
I use a taranis.

kh4, 2015-04-02:
@kilrah worth noting that frsky telemetry is supported also by openLRSng which can achieve range of 10km at ease on good conditions/antennas with telemetry.

romoloman, 2015-04-27:
If we are wishing to calculate distances for more then 3km then actual math is completely inaccurate. When I wrote that math i made the assumption that sin(a)=a also we did a approximated expansion of the cosine serie and furthermore that the the piece of earth limited by latitude a longitude lines could be considered almost a rectangle (due to small difference in longitude) this allow to simplify the math to be implemented on a atmega without FP coproc. 
But for distances of more than 4km especially at high values of latitude this is completely inaccurate.
I fear that implementing the real formula even with the taranix/9xr-pro processor could be of huge impact from a performance point of view.
The formula for two point on the ground for any coordinate is 
dist (A,B) = R * acos(sin(latA) * sin(latB) + cos(latA) * cos(latB) * cos(lonA-lonB)) 
where R is 6372,795477598 (standard geoide radius)
But we still need to sum vertical distance which is not the simple difference of the height as we should also consider that the two perpendicular are not parallel (but we can make an approximation on this point at least for RC usage)

projectkk2glider, 2015-04-27:
I would vote for an implementation that is still relatively CPU friendly and gives us good enough precision (lets say error of plus/minus 50m or so at 10km distance). @romoloman is this possible with current math or some other simplistic implementation?

romoloman, 2015-04-27:
I have to try to do some math.... I would like to avoid in any case the usage of sin / cos and acos functions especially acos function is very CPU intensive without a FPU unit...

bsongis, 2015-04-27:
I think sin(x) should be a little bit more precise (on ARM boards) than what we have now, but I do agree with Romolo on this, not too much maths!
I have to check, on X9E we have an FPU unit if I am not mistaken!

romoloman, 2015-04-27:
Bertrand actually is not the SIN the problem but COS that we use to calculate the radius from earth rotation axis at a give latitude.
If you remember how we calculate distance (the algorithm is a little bit hidden by your very good code optimizations) we calculate the distance between two different latitudes as Dy=|R*(lat1-lat2)|
then the distance between two different longitudes as Dx=|R*cos(lat1)*(lon1-lon2)| then we use Pitagora's theorem with dist=sqrt(Dx^2+Dy^2). This for horizontal distance, when we have also altitude we also add the Dz^2 term in the square root.

kilrah, 2015-04-28:
> I fear that implementing the real formula even with the taranix/9xr-pro processor could be of huge impact from a performance point of view.

There are several multirotor flight controllers running on atmegas with GPS navigation that run that kind of calcs several times per second along a lot of other lot of complex math and time-critical processes... can't see how it could be an issue on a fast ARM CPU like we have.

projectkk2glider, 2015-04-28:
@kilrah  you are absolutely right. Let'ts look at how the Arducopter does this. This looks like a very simple calculation: https://github.com/diydrones/ardupilot/blob/a9eec29e45bc6fa96c34e07b0c8c7ccedef43a61/libraries/AP_Math/location.cpp#L55

We might even consider changing the distance into horizontal distance only or add a new filed `horizontal distance`.

kilrah, 2015-04-28:
Not worth adding a field on master I'd say, especially as next already supports both 2D and 3D (choose both a GPS sensor and an alt sensor for 3D, if no altitude sensor is chosen you get 2D).

bsongis, 2015-04-28:
Yes we don't add a new field in 'master', it would mean translations etc.

romoloman, 2015-05-01:
About arducopter implementation: is exactly the same the same I propose with use of just COS() function but that means float, and float are no good for 9X, (cos() in my opinion is no good for Taranis/9XR as well) but I never tested how much does it takes)

amontefusco, 2015-05-09:
Could a CORDIC implementation of sin/cos be useful in order to reduce the load on the CPU  ?

http://www.dcs.gla.ac.uk/~jhw/cordic/

wilco1967, 2019-01-21:
I know this is an old item, but it seems like this issue still present in 2.2.3.

dist calculation seems to work fine upto about 4483 meter, and then jumps back to 1363 and start counting backwards. To me, looks like some kind of overflow.

Can provide a logfile if someone wants to have a look...

3djc, 2019-01-21:
Yes please, I’ll have a look

kilrah, 2019-01-21:
Since we don't need to care about AVR anymore we should be able to switch to a more precise but processor intensive algorithm without issue.

wilco1967, 2019-01-21:
Thanks guys for looking into this...
not really important, just a little annoying.... 

here's a google drive link with a 14 km max distance mission... where the 'dist' jumps many times....
[https://drive.google.com/open?id=1QXCD5Z69LhqwsTLAkwGxfb9nJGus4KT-](url)

3djc, 2019-01-22:
Thanks for your logs, we have a fix nearly ready

closed 2019-08-19 by kilrah