opentx/opentx #2176: Some sounds not played by Play Sound Special Function

projectkk2glider, 2015-03-29:
Some sounds won't play if the audio queue is not empty. Example:
[image]

First two SF produce desired result, SAup causes Cheerp sound then *gear down* is played. 

Second pair of SF does not produce desired result, only  *gear down* is played.

It is caused by the following condition in audio_arm.cpp (https://github.com/opentx/opentx/blob/master/radio/src/audio_arm.cpp#L978) that was there for a long time:
```c
void audioEvent(unsigned int index, unsigned int freq)
{
.
.

    if (index < AU_FRSKY_FIRST || audioQueue.empty()) {

```

This only affects sounds that have ID bigger than AU_FRSKY_FIRST:
```c
    AU_FRSKY_FIRST,
    AU_FRSKY_BEEP1 = AU_FRSKY_FIRST,
    AU_FRSKY_BEEP2,
    AU_FRSKY_BEEP3,
    AU_FRSKY_WARN1,
    AU_FRSKY_WARN2,
    AU_FRSKY_CHEEP,
    AU_FRSKY_RATATA,
    AU_FRSKY_TICK,
    AU_FRSKY_SIREN,
    AU_FRSKY_RING,
    AU_FRSKY_SCIFI,
    AU_FRSKY_ROBOT,
    AU_FRSKY_CHIRP,
    AU_FRSKY_TADA,
    AU_FRSKY_CRICKET,
    AU_FRSKY_ALARMC,
    AU_FRSKY_LAST,
```

Should we consider removing this limitation, or is there a justified cause for this limitation?

bsongis, 2015-03-29:
Yes this test has no sense now ... But the audioEvent has also no sense, we shouldn't go into a switch there, it's there because it's duplicated from AVR boards (where we saved flash doing things like this ...)

projectkk2glider, 2015-06-16:
>But the audioEvent has also no sense, we shouldn't go into a switch there

@bsongis  What do you mean? I don't see how we can remove the `switch` statement. Please elaborate.

bsongis, 2015-06-16:
It's just the play are called which is not nice, we call audioEvent and then there is this switch, it could be simplified ...