opentx/opentx #2183: Feature Request:  Failsafe “HOLD” option for individual channels

MIKE-DAILEY, 2015-04-04:
I’m new to OpenTX and just got my Taranis Plus last week and am running 2.0.15.  

I’m very impressed with OpenTX so far, but I’ve discovered that it does not have a feature that I always used on my Futaba 12FG.   

On the Failsafe screen it would be great if it could support the option to have individual channels have the “Hold” function as well as the existing preset functions.  This provides the best of being able to set each individual channel for the safest failsafe.  

Not sure the FrSky X8R receiver limits this feature.     

Thanks,

Mike

MIKE-DAILEY, 2015-04-15:
Any thoughts on the possibility of the X8R supporting this feature?

Mike

scoopdreams, 2015-04-15:
Just don't set it and it'll hold to the last known signal? Isn't that how
it works?

MIKE-DAILEY, 2015-04-15:
I'm running 2.0.15 and once you select the "SET" option screen it moves the servo to whatever position is "SET" on the screen.  So if you do not change and servo it shows 0.0 and that is where it goes on failsafe.  On my Futaba 12Fg it has a screen like OpenTX, but each servo has an option for "SET" or "HOLD".

Mike

romoloman, 2015-04-27:
This request should be placed as first to FrSky, there is no way in the X8R,X6R and X4R to set a failsafe that way.
It's not an opentx issue but a hardware one.

kilrah, 2015-04-27:
AFAIK the hardware already supports it, it is an OpenTX issue as in we've never made use of it.

bsongis, 2015-04-27:
Yes, I should work on this issue, this is something missing.

MIKE-DAILEY, 2015-04-29:
There are a number of ways this feature could be implement, but my suggestion would be to add a "hold" check box after each CHX number and before X.X setting number on the Failsafe screen.  Checking the box would make that channel a hold.  The unchecked the setting would use, either 0.0 or whatever was set by the user.  Or is could be reversed, so unchecked was "hold' and checked would use the setting.      

[image]

Thank you very much for considering my request to be included in the build process.

Mike

closed 2015-05-11 by bsongis

MIKE-DAILEY, 2015-05-11:
Would it be possible for one of the developers post a screen clip of what the new Failsafe screen looks like?  I've looked through the code and I'm not able to visualize how it looks.  

Thanks,

Mike

bsongis, 2015-05-12:
Why don't you try the nightly builds?

MIKE-DAILEY, 2015-05-12:
Thanks for that suggestion Bertrand and I managed to do it and tried out the feature. It works really neat!  You folks are amazing and thank you for all of your work and contributions you make for us to have the coolest radio systems on the planet.

Mike
[image]

commit 287b6cdb (2015-05-07): Fixes #2183
commit 6668db72 (2015-05-10): Fixes #2183
commit 059ea49f (2015-05-11): Fixes #2183
Failsafe settings on individual channels