opentx/opentx #2184: telemetry

erieqc, 2015-04-04:
You guy's are doing a great job and I want to thank you for that. As long as you are redoing the telemetry instead of having one rpm sensor capability people with twin engines will be asking for the ability to have two rpm readouts. Just a thought.
Thanks Bill

bsongis, 2015-04-04:
You will be able to have 30 motors in 2.1, please have a look to our latest nightly builds!

closed 2015-04-04 by bsongis

erieqc, 2015-04-04:
Ok thank you.

Sent from Windows Mail

From: Bertrand Songis
Sent: ‎Saturday‎, ‎April‎ ‎4‎, ‎2015 ‎5‎:‎03‎ ‎AM
To: opentx/opentx
Cc: William Simon

You will be able to have 30 motors in 2.1, please have a look to our latest nightly builds!

kilrah, 2015-04-04:
See here http://openrcforums.com/forum/viewtopic.php?t=6887