opentx/opentx #2187: "Copy model" should warn when using same receiver number

zieren, 2015-04-05:
I'm using a DIY DSMX module. When I copy a model the receiver number is copied to the new model slot. But it's the same number as in the original, which will have confusing effects since (AFAICT) the module stores settings keyed by receiver number and will overwrite the original's settings when binding for the copy. It would be nice to show a warning or even offer to pick an unused number.

griebd, 2015-08-25:
Shouldn't we do the same warning on companion?

zieren, 2015-08-29:
That sounds like a good idea.

kilrah, 2015-08-29:
There are also other adjustments needed... When checkModelId() was written there was only one ID for both modules, now we have separate ones for both. I've noticed the warning comes up even when a module is disabled, which makes little sense. Might be better to skip the check if the module is set in a mode that doesn't use the ID, and rather do the check when said module is enabled.

griebd, 2015-08-29:
I think I got your point @kilrah (still getting acquaintance with the code), I will look better at when to checkModelId()!
About the separated ids, I knew about then in 2.1 but choose to ignore then for now! Why? Because this way I get knowledge of the code and solve the problem in 2.0. Later on I focus on 2.1 and make the changes to take in account the separated ids.
About companion, will let it for later too. One thing a time.

kilrah, 2015-08-29:
Ah yes, that only targets 2.0... all good then, nothing more needed.

griebd, 2015-09-17:
The warning about duplicated ids is already present in 2.1...
BUT apparently there is a bug! It does not take in account the separated ids for external and internal modules. (at leas on the eeprom converted from 2.0)

griebd, 2015-09-30:
OK, I need help from the experts here.... :wink:

From my understanding of the code when checkModelIdUnique() is run I have access to g_model (the current loaded model) and modelHeaders (that has all existing model names with their respective module ids). Am I correct so far?
With that in mind I have already fixed the but I mentioned in my last post.
BUT: If a module is disabled or make no use of the ID (because of the selected protocol) the check is still done!
Taken my understanding above, am I right to say there is no simple way to check the protocol of the other models to decide if the check is need or not?

bsongis, 2015-09-30:
You are right. Only the ModelHeader structures are stored in RAM, not the module settings!
By the way this feature will be a problem on new radios such as Horus which won't have EEPROM, only SD. Unless we open all models to check if the ID is used or not (we could also do that with EEPROM)

griebd, 2015-09-30:
I go for doing it! It's something that is not used all the time, so no problems with the resources (and run time) usage when it's called and it can really avoid real problems with models left on by mistake...
Going even deeper on that topic, in companion only (or radio too) we could make a check to see if there are any ids used twice.
What are your opinions? If you think it's useful I will invest the time and do it (from what I have learned so far it shouldn't be too big a task).

bsongis, 2015-09-30:
Yes this feature is definitely useful. With your change, it will even be possible to save some RAM (120bytes) in 2.2 in removing the 2 IDs from the ModelHeader.

griebd, 2015-11-03:
after some delay I'm back... and with new doubts!!!
I'm a bit lost, how should I load a model temporarily? I mean, the standard functions only load a model to the g_model global variable, right?
I'm trying to write a parallel loading function to do the same but to a local variable... question is:
What will be really needed for this?
What are the objectives of eeprom_common, eeprom_raw and eeprom_rlc?
Thanks!!!

bsongis, 2015-11-03:
You should only read the first bytes of the ModelData (only the header). Have a look to the function which loads all model headers, you will see the code already there.

griebd, 2015-11-03:
But on the model header I have no info about the protocol... I want to check only the models that use the same protocol type, no?!
In my case: I have a DIY DSM module on my taranis, I only want to check for duplicated ids on other models that use it too. If I get another module for another protocol, then those models can use the same id, I don't care. Am I wrong here?

griebd, 2015-11-03:
Something like this:
```c++
void checkModelIdUnique(uint8_t index, uint8_t module)
{
  uint8_t modelId = g_model.header.modelId[module];
  if (modelId == 0) {
    return;
  }

  uint8_t type = g_model.moduleData[module].type;
  if (type != MODULE_TYPE_DSM2 && type != MODULE_TYPE_XJT) {
    return;
  }

  if (type == MODULE_TYPE_XJT) {
    int8_t rfProtocol = g_model.moduleData[module].rfProtocol;
    if (rfProtocol != RF_PROTO_X16 && rfProtocol != RF_PROTO_LR12) {
      return;
    }
  }

  ModelData g_model_tmp;
  for (uint8_t i=0; i<MAX_MODELS; i++) {
    if (i != index) {
      eeLoadModelTmp(i, (uint8_t *)&g_model_tmp, sizeof(g_model_tmp));
      if (type == g_model_tmp.moduleData[module].type) {
        if (modelId == g_model_tmp.header.modelId[module]) {
          POPUP_WARNING(STR_MODELIDUSED);
          return;
        }
      }
    }
  }
}
```

bsongis, 2015-11-03:
Oh ok understood, and no it means you have to download the whole model, it's not optimal!

griebd, 2015-11-03:
No, I agree, but it's done only when creating a new model or a protocol is changed to one that has ids or an id is changed. That means this code will rarely by run, but its function will help to safety.

bsongis, 2015-11-03:
Yes, it's worth a try, I am just afraid it will be slow to load all models to check the model ID

bsongis, 2015-11-03:
Another ID that could be done in 2.2: store the module settings at the beginning of the structure.

bsongis, 2015-11-03:
Later ... it would be nice to try the new sdcard storage interface on Taranis (from my Horus branch)

kilrah, 2015-11-03:
> or a protocol is changed to one that has ids or an id is changed.

Note that with a nice big eeprom this check will take several seconds to run. So if you have a model with ID 1 and want to change it to ID 10, you'll have to press "+" 9 times, and have to wait 5 seconds after every press so that the checks run? ;)

bsongis, 2015-11-03:
It could be done when edit is done (as we do for the radio mode)

griebd, 2015-11-04:
Need to check, but I think the check is done when you "exit the field", when you press enter or exit, not on every change.

Johnex, 2017-06-07:
So since we all got sent here, is there any update to how this could be changed to avoid confusion?

PR #2770 (2015-09-13): Alert about duplicated receiver id on copy. #2187
Fixes #2187.

Again, review with care and point me in the right direction.
commit 3018328a (2015-09-13): Fixes #2187. Warning about used receiver id on copy.
closed 2019-11-15 by stale[bot]