opentx/opentx #2188: Add support for replacement LCD ERC12864-2

universam1, 2015-04-06:
Assume there is a general interest in a good replacement for the LCD of the 9X & 9XR - had a hard time to find a fitting substitue. I've no interest in advertising, just if someone needs it's almost same size 2,6", hardware compatible (backlight needs pin swap) with IMHO better quality. Search for ERC12864-2 in eBay or buydisplay.com

Unfortunatelly with current init its not usable, almost no contrast. After lots of tests the manufacturer explained that the init sequence needs some changes:

"We checked your initalizer,a few instructions need to be modified,See below please;
0x25, //Select int resistance ratio R2 R1 R0 =5_____Modify 0x25 to 0x27
0x22, // 24 SV5 SV4 SV3 SV2 SV1 SV0 = 0x18_____Modify 0x22 to 0x2D
g_eeGeneral.contrast = 0x22;______Modify 0x22 to 0x2D
Bias(0xA2, // Select LCD bias=0____Modify 0xA2 to 0xA3)."

With these changes it works excellent. Would love to get general support here!?

[image]

bsongis, 2015-04-06:
Yes, with an option in Companion, would you provide the patch?

kilrah, 2015-04-06:
From your photo that would be only for 9XR, right? As far as I know the LCDs are not interchangeable between 9X and 9XR due to a different connector.

universam1, 2015-04-06:
Connector actually is the same, I've tried it on a 9X. Unfortunately I
don't have access to a 9X any more, so a schematic would be needed. Anyone?

kilrah, 2015-04-06:
I think the connector is the same but mirrored (i.e. the LCD would end up facing the board unless you twisted the flat cable, which you can't).

universam1, 2015-04-06:
Likely, how should the define be named?

universam1, 2015-04-06:
Thats true it was mirrord! Though, theoretically, the cable is 6cm long
while the LCD 5cm is shorter, you could wrap it the long way, but then the
display would be upside-down.
Or I just tried to twist the cable, that means I folded it 2x 90° which is
okay since the cable is long enough, and it still shines at me ;-)

universam1, 2015-04-07:
buydisplay.com:
"Here is our answer about the below questions;

1.This module ERC12864FSF-2 is the one of our popular product,It will be keep on sale at recent years,We will inform the customer in advance if we plan to upgrade this module,So please don't worry about it.

2.Our ERC12864FSF-4 could reach your requirement,It has same 30Pin connector and same programd as our ERC12864FSF-2,only difference is the voltage,Please refer to the datasheet for more details at http://www.ebay.com/itm/12864-128x64-GLCD-Graphic-LCD-Module-Display-SPI-Serial-ST7565P-3V-Black-White-/291022273441?pt=LH_DefaultDomain_0&hash=item43c2477ba1
"

projectkk2glider, 2015-05-05:
The option for `LCD_ERC12864FSF` must be added to the Makefile, then this issue can be closed.

bsongis, 2015-05-21:
Done

closed 2015-05-21 by bsongis

tedbmoss, 2015-07-09:
This is a very good idea.
Thanks