opentx/opentx #2190: Ability to display current time in telemetry screens

ckleanth, 2015-04-08:
Not sure if this was addressed/can be done with the new telemetry revamp in OTX 2.1 but currently with OTX2.15, the current time disappears once you enter the telemetry screen menu.

One might say use a watch or your phone to check the time or exit the telemetry menu but hey if it can be added or can be done it will save me an argument of being late going back home…

kilrah, 2015-04-08:
You can already do so, and even set an alarm.

[image]
[image]

closed 2015-04-08 by kilrah

ckleanth, 2015-04-08:
Cool thanks Andre

kilrah, 2015-04-08:
And that's on 2.0.15 I might add.

ckleanth, 2015-04-08:
Yes typo, sorry about that