opentx/opentx #2193: getValue ("timer1") and getValue("clock") not returning value on Taranis

RuKeBo, 2015-04-11:
A similar issue was opened and closed here: https://github.com/opentx/opentx/issues/2037.  This issue had to do with getValue ("timer1").

This was being discussed on RCGroups (http://www.rcgroups.com/forums/showthread.php?t=2180477&highlight=lua+simulator&page=82) and I am having the same issue.  I don't get any reading on the Taranis with getValue("clock") or getValue(190), also clock.  There are other ways to get the Timer1 value, none for the clock as far as I know, plus it seems like it should work.  As mentioned in the other issue, the does work in the simulator, just not in the Taranis.  It has also been confirmed by another person.

projectkk2glider, 2015-04-12:
I will check this one.

projectkk2glider, 2015-04-12:
Confirmed, works in simu, but not on the radio.

projectkk2glider, 2015-04-12:
Sample script:
```lua
local function run(event)
  lcd.lock()
  lcd.clear()

  local t1 = getValue("timer1")
  local t2 = getValue("timer2")
  local cl = getValue("clock")

  print("t1: "..t1.." t2: ".. t2.." clock: "..cl)

  lcd.drawText(0, 10, "t1: "..t1.." t2:"..t2, 0)
  lcd.drawText(0, 20, "clock: "..cl, 0)

  return 0
end

return {run=run}
```

All three values have value of zero on the radio all the time.

projectkk2glider, 2015-04-12:
It's a simple explanation:
* the values `timer`, `timer2` and `clock` are considered by the radio as the telemetry values (which I think is actually wrong).
* Lua function `getValue()` reports zero value for all telemetry values if the RSSI is zero - ie if telemetry is not being received.

This piece of code is to blame:
```c
static void luaGetValueAndPush(int src)
{
  if (src >= MIXSRC_FIRST_TELEM && src <= MIXSRC_LAST_TELEM) {
    // telemetry values
   if ((src != MIXSRC_FIRST_TELEM-1+TELEM_TX_VOLTAGE) && !TELEMETRY_STREAMING()) {
      // telemetry not working, return zero for telemetry sources
      // except for "tx voltage"
      lua_pushinteger(L, (int)0);
      return;
    }
  }
```

The idea of returning 0 for all telemetry values when RSSI iz zero was my, done in https://github.com/opentx/opentx/commit/01391aa6aacf9e84a8f2f9fd49676e28a0d1bf53 but I can not remember why I thought it was such a good idea to implement.

Now how do we get out of this hole? If we remove this condition and always return values it will affect (some) current scripts/users. @bsongis  @kilrah

projectkk2glider, 2015-04-12:
I just remembered why I implemented `return 0 if RSSI == 0`. That was way back when RSSI value was not available to Lua so scripts had no way of knowing if telemetry was streaming or not. So I decided to return 0 for the few telemetry values that we had then.

Then we implemented the Lua fields which exposed (all) telemetry values and this  `return 0 if RSSI == 0` is now really obsolete.

Still, if we remove it, I am sure some user scripts will be affected.

bsongis, 2015-04-12:
I vote for a removal!

RuKeBo, 2015-04-12:
I am confused why RSSI is zero if the radio is connected to the receiver and receiving data.  Shouldn't RSSI be anything but zero and it return the telemetry values?  Also, if previously RSSI was not available, then when would it be anything but zero and when you the values be returned?

kilrah, 2015-04-12:
But did you and the other guy reporting the issue really have the receiver bound and powered when you tested? Sounds like you just put the script on and tried it without a running receiver, and in this case the behavior would be "normal" at this point.

projectkk2glider, 2015-04-12:
@RuKeBo RSSI is zero if telemetry is not received. You can not have a situation where you have an RSSI value of zero and at the same time you are receiving other telemetry values. 

I suggest you try the same script that was giving you only zeros while you have an active connection with properly bound receiver - ie working telemetry. Then you should get the values for timers and clock. Test and report back.

RuKeBo, 2015-04-12:
I guess I was assuming this was being done with receiver power on, which might not be true. 

Well this is strange.  I know at one point I powered the receiver and was looking at the telemetry, I just don't recall if I noticed if the clock was reading correctly or not.  So per your instruction and since I wasn't sure, I just went to test it with the receiver powered on.   Before I did that, I simply turned on the Taranis (no receiver power) and for some reason, the clock was showing the correct time without any RSSI value, the RSSI was zero.  I didn't change anything and now it is working with the receiver not powered on and contrary to how you just explained it should work, so I don't understand why.  I will need to do more testing and get back to you.  I have a switch call out the RSSI value and also have it displayed on the screen and both indicate it is zero.  I thought maybe it might be seeing something other than zero, maybe 0.0001 fro example, while the receiver unpowered, but it says it is zero.  Right now it seems to be working both ways.  I will copy your script in and test that although the code is basically the same.

RuKeBo, 2015-04-12:
Disregard the above.  I did change something in my code.  I used the lcd.drawChannel() function instead of getValue() which is why it was showing a value. I will try it with the getValue() function and receiver powered and get back yo you.

RuKeBo, 2015-04-12:
Ok, I can confirm this works as you say.  I used your script above, turned on Taranis, it displayed all zeros.  I plugged power into receiver and values registered.  Thank you for explaining this. I am not sure this is how it should work, but it is working how you said it should and how it was programmed.

kilrah, 2015-04-12:
Good!

I'd vote for no change in master to avoid breaking script compatibility, anyway when testing a telemetry script it's you'd want telemetry to be running in the first place. Just testing the clock standalone isn't a very common scenario...

RuKeBo, 2015-04-12:
Ok.  I was not testing the clock, I was just displaying the time.  Surely not a critical thing to need to display, but since it didn't work, I was wondering why, now I know.  It sort of affected the timer as well which was something I wanted to display. Thanks.

closed 2015-04-12 by RuKeBo