opentx/opentx #2194: ASSISTANT HELICO

ventilo76, 2015-04-11:
Bonjour,

Je viens d'acheter une Taranis plus, et je viens de m'apercevoir que l'assistant dans OpenTX (l'application sur le PC)  est grisée et donc inutilisable...
Pourriez-vous le rétablir et est t'il possible d'avoir le décalage virtuel sur le plateau cyclique (maquette avec attaque direct sur pied de pale - absence de barre de bell) En général le décalage se situe entre 42 et 45°, le mieux serait de pouvoir le régler nous même.

Merci d'avance.

bsongis, 2015-04-11:
Ben voyons ... qui va faire ce travail là pendant son week-end ? Je crois qu'il va falloir acheter une radio Futaba, leur appli pour PC est top ;)

kilrah, 2015-04-11:
L'assistant hélico n'a jamais existé, sa place a été préparée mais personne ne l'a jamais fait. 
C'est volontiers qu'on incorpore le code de quiconque veut s'y lancer.

ventilo76, 2015-04-11:
Tout d’abord, BONJOUR !

Je pense que dans toute communauté il faut commencer par cela, un minimum de politesse ne peut qu’améliorer les relations.

 

Je ne vous connais pas et je suis surpris par votre réponse …

Pouvez-vous me faire une copie écran où j’aurais demandé quelque part que ce soit fait pendant le WE ?

 

Je pense que c’est suite à un message que j’ai posté à un endroit où tout est anglais sur les conseils donnés sur ce forum …

 <http://frskytaranis.forumactif.org/t1523-assistant-helico#23764> http://frskytaranis.forumactif.org/t1523-assistant-helico#23764

 

A part cela je n’y connais rien en votre système ou communauté, j’ai acheté cette radio car parait t’il c’est la radio du siècle…

 

Merci pour le conseil, j’ai déjà deux 1024 de Futaba haut de gamme, une avec programme avions et l’autre programmes hélicos avec tous les plateaux  existants (mais sans le décalage virtuel).

En intégré, elles ont les voltmètres ainsi que compte tours optique.

 

Cordialement,

 

Philippe DUVAL

 

PHD 3.jpg

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
http://www.avast.com

ventilo76, 2015-04-11:
Tout d’abord, BONJOUR !

Je pense que dans toute communauté il faut commencer par cela, un minimum de politesse ne peut qu’améliorer les relations.

 

De quels codes parlez-vous ?

 

Qu’avez-vous besoin ?

 

Cordialement,

 

Philippe DUVAL

 

PHD 3.jpg

 

http://www.avast.com

kilrah, 2015-04-12:
>De quels codes parlez-vous ?
Qu’avez-vous besoin ?

Que quelqu'un implémente la fonctionnalité... On a actuellement 222 requêtes en attente, l'assistant hélico en est une parmi les autres et ne semble pas être une priorité pour les contributeurs actuels. Comme un petit nombre de volontaires faisant ça dans leur temps libre sans rien demander en échange ne peut pas tout faire à la fois, il faut faire des choix et les requêtes qui sonnent comme un peu trop insistantes sont souvent ressenties comme impolies... bien plus qu'une absence de "bonjour" ;)

La requête pour l'assistant hélico existe déjà ( #906 ), je ferme donc celle-ci qui fait doublon.

kilrah, 2015-04-12:
>De quels codes parlez-vous ?
Qu’avez-vous besoin ?

Que quelqu'un fasse le boulot et implémente la fonctionnalité... On a actuellement 222 requêtes en attente, l'assistant hélico en est une parmi les autres et ne semble pas être une priorité pour les contributeurs actuels. Comme un petit nombre de volontaires faisant ça dans leur temps libre sans rien demander en échange ne peut pas tout faire à la fois, il faut faire des choix et les requêtes qui sonnent comme un peu trop insistantes (comme EN CRIANT EN MAJUSCULES par exemple) sont souvent ressenties comme impolies... bien plus qu'une absence de "bonjour" qui lui est plus ou moins considéré comme obsolète ;)

La requête pour l'assistant hélico existe déjà ( #906 ), je ferme donc celle-ci qui fait doublon.

C'est tout le principe de l'open source, chacun est libre de participer et de travailler sur ce qui l'intéresse.

closed 2015-04-12 by kilrah

ventilo76, 2015-04-13:
Bonjour,

 

Ah je vois, dommage je n’ai pas les compétences requises pour participer…

 

Mais pourquoi tout est en anglais sur votre site ou forum ? Êtes vous un des rares français ?

 

Cordialement,

 

Philippe DUVAL

 

http://www.avast.com

kilrah, 2015-04-13:
Les 2 principaux contributeurs sont francophones... Mais 95% des utilisateurs sont internationaux. On a déjà de la peine à garder la doc à jour en une langue, alors 2...