opentx/opentx #2196: Critical: "TX Antenna Problem" error on FrSky Taranis Plus

MIKE-DAILEY, 2015-04-11:
There seems to be an extremely nasty problem being reported with the newer Taranis Plus radios that have the new main board with the XJT on the board that provide the standing wave measurement.  The transmitter will intermittently display "TX Antenna Problem"  followed by voice "Signal Loss" and then crash!  This is a highly intermittent problem. 

Could this just be a glitch in the standing wave calculation, and the most important and critical question, does 2.0.15 shut down the RF on this error.  The current Taranis Plus radios come from the factory with 2.0.9.2 and apparently it does shut down the RF.

There is the known glitch on getting the "TX Antenna Problem" when doing a telemetry reset, but on my system running 2.0.15 during the error on reset, the RF is not shut down. 

Help, this is getting to be a huge problem and snowballing.

Thanks,

Mike

bsongis, 2015-04-11:
Is there anything new on this topic? The RF is NOT shutdown on Telemetry Reset. It seems to me that the user really had a TX antenna problem which the firmware cannot fix.

MIKE-DAILEY, 2015-04-11:
Thanks for your very quick response.  So, just to be clear, you are saying that with 2.0.15 the RF is not shut down if there is a "TX Antenna problem" error pop-up during normal flying operations?

Thank,

Mike

kilrah, 2015-04-12:
At least OpenTX does not stop transmitting on an antenna problem, all it does is display the warning. 
The module however may do what it wants and could stop, but from out tests it is not the case. With current module firmware when causing the antenna problem on purpose control is maintained.

So if there's a problem somewhere there isn't anything we can do about it on our side, it's more likely to be a hardware issue.

MIKE-DAILEY, 2015-04-12:
Thank you very much for your response.   Looks like FrSky has a problem with radios made in the last few months.    

Mike

bsongis, 2015-04-12:
How many crashes like this have been reported? There was one report on openrcforums, did anybody else experienced the same issue?

tomkfly, 2015-04-12:
There are a few reported on RC Groups. Not a lot, but enough to make it worrying.

hrenz, 2015-04-12:
Same reports at FPV-Community
Antenna problem accours during flight with TaranisPlus Prod-date Dez2014
Or after 5 or 10 times Telemetry Reset an Antenne problem accours too
with 255 and then slow go down to 0 or 1

mpjf01, 2015-04-12:
A separate thread has now been set up in RCG to collate reported instances of antenna warnings on Taranis in the interests of getting all the reports on that forum in a single place. They are spread between four of five threads at present..

http://www.rcgroups.com/forums/showt...7#post31322989

kilrah, 2015-04-12:
The warning when telemetry is reset is unrelated and solved.

bsongis, 2015-04-15:
I was given 2 different information yesterday, not sure both together make sense, and difficult to know today if it is an hardware or a firmware failure. For the moment the problem seems in the module though.
1) flashing the module with the previous firmware version removes the problem
2) only radios "B02" with the IPX connector have this problem

bsongis, 2015-04-15:
By the way if your radio raises an "Antenna warning" on the bench, I wouldn't fly with it.

MIKE-DAILEY, 2015-04-15:
The main thing I wanted to find out is if OpenTX 2.0.15 shuts down the RF on the "TX Antenna Problem". I believe the answer to that at this point is NO.  I have a March 11, 2015 B02 system with 2.0.15 and IPX connector from Aloft Hobbies and after 25 hours of operation I have not had any problems at all.  There is a reported single event on a system with a serial number that is about 300 numbers older than mine so I did a lot of bench running before I flew it.  

Aloft Hobbies is working with FrSky to determine what is causing the problem.  When the reports of the problem first started being reported it looked like it could be wide spread because one person had two systems with the problem (one without the IPX and one without) etc., then after about 5 or 6 reports there haven't been any more on RCG. One of the systems did not show any problems, but then had an event after 15 hours of run time.  

So it looks like it is limited to very few systems and the problem is very intermittent.  There is a report of a system with 2.0.13 that had a event without loss of control so that is good news because that was not the case with the factory installed 2.0.9.2. 

Hopefully Aloft Hobbies and FrSly will get this figured  out so we all know the root cause and we can fly without any worries. FrSky will be able to analyze the firmware in the internal XJT RF module and find the root cause.

bsongis, 2015-04-17:
Latest news ... the problem seems to be related to the IPEX connector. A wrong batch.

MIKE-DAILEY, 2015-04-17:
Does anyone know the serial numbers or build date of the radios impacted?

bsongis, 2015-04-18:
This is something to ask to FrSky! By the way I think we can close the issue, it has nothing to do with OpenTX, and we can't fix anything at our level.

closed 2015-04-18 by bsongis

sandrinio01, 2016-01-31:
can someone help me PLS. I have taranis 9xd. when the range is more then 120 the alarm starts at 45 RSSI. how can i increase the range. as I know it should work in more then 2 km range. what can be the  problem reason?

projectkk2glider, 2016-01-31:
Please use forums for such questions. Just to get you started, try with another receiver, if the range is the same with the new receiver, then the radio has a problem.

sandrinio01, 2016-01-31:
i dont have another receiver. is there any other way to test it?

projectkk2glider, 2016-02-01:
>is there any other way to test it?

No, unless you have access to expensive RF measuring instruments.

DjLiviuX, 2017-02-22:
hi, i am writing this for everyone to know and understand the risk of modding the taranis antenna.
If you mod the taranis tx module with and external antenna be aware not to touch the live wire ! The final IC that does the TX amplification gets ruined by ESD so be cautious !

kilrah, 2017-02-22:
Yep - between the fine soldering and high sensitivity of the PA to ESD it's really not recommended. Wayne at Alofthobbies says he sells multiple replacement boards per week to people damaging their radios trying to do the mod.