opentx/opentx #2197: error 2 no such file or directory when starting Sim on Companion 2.0.15 Ubuntu

jermoo, 2015-04-12:
f_opendir(/media/jermo/Data/SDCard//SOUNDS/en/SYSTEM) = error 2 (No such file or directory)

I have the SD card files from TXU copied and all other sounds appear to be found based on the terminal output while running the SIM. I reviewed the sim code file but can't determine if I'm missing an audio file or it's an actual issue. 

everything builds and appears to run correctly otherwise. 

Just letting it be known. I'm starting work with LUA to see what I can accomplish.
Jer

projectkk2glider, 2015-04-14:
Try this command in terminal window and copy back the result:
```
stat /media/jermo/Data/SDCard//SOUNDS/en/SYSTEM
```

My bet is that this path really does not exist, or you have a permission problem.

kilrah, 2015-04-14:
There are 2 slashes in there. Probably should remove the trailing one in the SD card structure location string..

jermoo, 2015-04-16:
ok.. the stat command failed. The folder was lower case 'system'. Renamed to SYSTEM, stat now passes, however Companion persists in the error. It MAY be permissions based so let me try to chown the files and see if that works.

jermoo, 2015-04-16:
Resolved, this was a combination of case of the folder and permissions. The Companion doesn't generate a warning and surprisingly now makes sound.
User error. Thank you for the help.. now if I can just figure out how to work on code with Github so I can start contributing.

closed 2015-04-16 by jermoo