opentx/opentx #2198: wrong selection when choosing a fly sessions in log viewer

griebd, 2015-04-13:
Just (actually many days ago :wink:) noticed that when you select a fly session in log viewer the selected rows doesn't mach the fly session.

The problem is that some variables are declared as int when they should be double.

Will make a pull request soon (my last pull request to next already included the changes).

griebd, 2015-04-21:
Fixed on next by #2191!

bsongis, 2015-04-21:
Would you fix it on 'master' as well? Or it's just too much work and doesn't need to be fixed in 2.0.16?

griebd, 2015-04-21:
it's already fixed and merged on master, #2200. The issue was still open because of next, but I'm preparing a new related pull request and noticed it's fixed and merged on next too.
The important part is, this is solved and can be closed. :wink:

taking the opportunity, should I leave it open for you guys to close or can I close the issues I solve? (yes, I will try to keep contributing to this project, I use it and find it easier to contribute codding than donating)

bsongis, 2015-04-21:
Yes you may close the issues you fix! Don't forget to update the milestone if needed. Thanks for your help!

closed 2015-04-21 by griebd

kilrah, 2015-04-22:
Issues close themselves automatically once the fix is merged on next if you reference them with "fix #xxxx" in the commit or pull request :)

PR #2200 (2015-04-13): [companion][log viewer]fix fly session selection
fixes #2198