opentx/opentx #2203: Telemetry graphic bug

LapinFou, 2015-04-14:
Hi folks,

There is a graphical bug with the French version of OpenTX v2.0.15 as shown below.
[image]

By the way, there is a restriction for RPM field. The maximum value which can be displayed in the bar graph is 12750. The propeller of one of my plane can reach 15000rpm. This higher RPM value is correctly record in the LOG and also correctly displayed in the classic telemetry screen. Maybe I should open another ticket?

kilrah, 2015-04-14:
Wow... did we finally find the first person who uses these bars?

bsongis, 2015-04-14:
I think we won't fix the RPM problem, it's completely changed in 2.1 and the restriction is not there.
Perhaps you could have a check of the bars in 2.1, it would help us!

bsongis, 2015-04-14:
I will reduce the bars length a little bit in 'master' before 2.0.16

kilrah, 2015-04-14:
Where do those strings come from? Could just use the 4-char ones

bsongis, 2015-04-14:
True, perhaps it's the translation which is wrong

closed 2015-04-14 by bsongis

LapinFou, 2015-04-14:
I think the bars display are very handy for reading remaining fuel, total consumption and RSSI.
No need to read the value, a quick look and the bars gives you the wanted information.
For RPM, it doesn't matter.
As soon as I will have more time, I will try the 2.1 beta version.

Thanks again for your hard work !
:)

commit 905e3c78 (2015-04-14): Fixes #2203
commit 3d59de9f (2015-04-14): Fixes #2203