opentx/opentx #2208: Telemetry issue when set limite for A2

blairfancy, 2015-04-18:
Hi i m not able to setup properly A2 limit for warning.

Parameters on A2;
gain: 332°
offset: 0
Alarme Orang>95°
Alarme Red>105°

in this configuration critical alarm is always played by radio

projectkk2glider, 2015-05-02:
The low and critical alarm work when A2 voltage is **lower** than particular warning voltage, not when higher as you wrote.

So in your case if A2 is less than 105 you will get a red alarm. The orange warning will never sound since it is overridden by higher setting of red alarm.

kilrah, 2015-05-02:
He is on 9XR Pro and on all "small screen" platforms the UI still allows choosing smaller or greater as a choice for both alarms, so it IS a valid issue.

Now how we want to fix it is a different matter, I would vote to unify with the Taranis as things like OP wants to do can be done with a logical switch/special function pair.

projectkk2glider, 2015-05-02:
Ups, I did not know that. You learn something new every day :)

blairfancy, 2015-05-02:
Thank for your comments, are there any other way to setup warning for telemetry. You speak about Switch/Logical function.
regards

kilrah, 2015-05-02:
Yes, that's the "standard" way to set up alarms. Use a logical switch e.g. a>x Temp 105°, then use that switch to trigger a "Play Sound" or "Play Track" special function.