opentx/opentx #2221: LCD contents shifted upwards by 6 pixels (with wraparound)

xmxwx, 2015-04-23:
'next' branch, 41f78440db2230a20bb980c54640c406a65e36cf

bsongis, 2015-04-24:
What? Why did you mention this commit, it is related to Companion! Did you compile your firmware yourself? 

We introduced options for LCD, I guess there is either an error in the Makefile or in the compile script

bsongis, 2015-04-24:
No I don't see any issue in the Makefile. Did you try many nightly builds to help me and find which commit is involved?

kilrah, 2015-04-24:
And is it on the actual radio or in the simulator?

xmxwx, 2015-04-24:
When I was working on https://github.com/opentx/opentx/pull/2210 I did everything on the master branch. @bsongis, when you suggested the right thing - to do it on the next branch - I've rebased my changes, compiled and flashed into 9XR and saw this problem. Then I checked out clean next (hence the mentioned commit) and saw that the problem is still there even without any of my changes. Thus, I posted this issue here.

@bsongis 
> Why did you mention this commit, it is related to Companion!

It is just the commit I've compiled and flashed into my 9XR. Not a commit that I think is bad.

> Did you compile your firmware yourself?

Yes.

> Did you try many nightly builds to help me and find which commit is involved?

No, not yet. I hope I'll have some time for this today in the evening. I'll try nightly builds. Or maybe I'll use my favourite git bisect because it will help me to do it logarithmically ;-).

@kilrah 
> And is it on the actual radio or in the simulator?

Actual radio. Frankly, I haven't yet tried the simulator.

xmxwx, 2015-04-24:
BTW. I firstly marked it with "9X-Stock (m128)" label, but then I'll noticed that it's "9X", not "9XR". Shouldn't we have "9XR" label?

bsongis, 2015-04-24:
Oh ok, now I understand everything, I didn't see it was the same person who opened this issue and the pull request!

bsongis, 2015-04-24:
Would you do some dichotomia to see which commit is wrong? 

git reset --hard origin/next~200
=> check

git reset --hard origin/next~500
=> check

etc.

xmxwx, 2015-04-24:
> Would you do some dichotomia to see which commit is wrong? 

Of course, that's what git bisect does :-)

Now I really gotta go... Till the evening!

bsongis, 2015-04-25:
I learnt something with git bisect, I was used to do it manually!

xmxwx, 2015-04-25:
I just finished git-bisecting.

c6cf0f915ec26f418d79afde6b9b4a7dc6bf819d is the first bad commit

bsongis, 2015-04-25:
> I just finished git-bisecting.
>
> c6cf0f9
> is the first bad commit

I see, would you revert it? I mean MEGA2560 should have its own
lcd_driver.cpp!

commit 8e074aaa (2015-04-26): Fixes #2221 (LCD contents shifted upwards by 6 pixels on 9XR).
commit b589cd62 (2015-04-26): Merge pull request #2228 from xmxwx/#2221

Fixes #2221 (LCD contents shifted upwards by 6 pixels on 9XR).
closed 2015-04-26 by xmxwx