opentx/opentx #2224: opentx 2.0+ companion save/import profile & settings to hard drive

ckleanth, 2015-04-24:
hi gents (and lasses)

Recently I had to change pc's and reinstall everything. The old companion9x v1.53 has a save/import profile which is missing from opentx 2.0+ companion (or at least I couldn't see this functionality.
I lost hardrive access from the old first pc so I couldn't dump the windows registry to reimport in the new pc.
HKEY_CURRENT_USER\Software\companion9x
HKEY_CURRENT_USER\Software\OpenTX

would it be possible to reintroduce the save/import profile as well as saving the whole companion settings without messing with the registry (for windows-dunno about other pc file structures)
thanks

kilrah, 2015-04-24:
I was the one who originally requested it, but in the end probably only used it once or twice.
Nearly quicker to just set them up manually again, and you get to fix/clean outdated stuff.

mpaperno, 2018-06-06:
Consolidating with #4996

closed 2018-06-06 by mpaperno