opentx/opentx #2226: Taranis with D4r-II shows A1 and A2 voltages, but only A1/Gnd is connected.

MagikMan4ever, 2015-04-24:
So i built a mini quad (emaxx 2204's, HQ 5045 props "Dumb", rotorgeek v4 esc's, the total weight with 154g 4 cell = 600g,) then crashed when i burnt up an esc, so i assumed those motors draw to much, everything else worked good as far as i know after, so i bought all cobra 2000's  and all 4 new esc's, with v13 bl-heli, and i will start with 5030 and move up later with v13 bl-heli, 

    Now i have it all built exactly the same as b4 Same 1.7.1 cleanflight firmware on naze32, same Taranis .15 Firmware, the only difference could be the openTx configurator is now 0.63.1 instead of 0.63.0.
   Just new motors and new esc's, and got it down to 625g's, calibrated esc's , has good balanced CG, and ready to test, But i'm afraid to fly Because..........

     Now my a1 and a2 telemetry values appear as voltages, when only the A1 showed b4. i am only using the rx and gnd "green and black wires" from the d4r-II to the naze32 telemetry pins just like before.
      Is this weird? could I have damaged my d4r-II in the crash? The A2 voltage doesn't seem to be in reference to to my battery voltage or my taranis voltage, and there is nothing accidentally touching the AD2 terminal on the d4r-II telemetry port.

projectkk2glider, 2015-04-24:
The A2 pin is floating and is picking up some leakage voltage/current. Just disable A1 and A2 in OpenTX telemetry setup if you don't use them (set both ranges to 0).

closed 2015-04-24 by projectkk2glider

hrenz, 2015-04-26:
Not used analog-inputs should be set to Ground.
This is a basic principle  for all AD converters